jeelabs/esp-link #490: esp-link crashing after bcn 100 command

codehero, 2020-04-27:
after flashing v3.0.14 my ESP8266 constantly resets with a "wdt reset" or "normal" error. I have ensure correct flash map and speed.

```
ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 1856, room 16 
tail 0
chksum 0x63
load 0x3ffe8000, len 776, room 8 
tail 0
chksum 0x02
load 0x3ffe8310, len 552, room 8 
tail 0
chksum 0x79
csum 0x79

2nd boot version : 1.5
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

...
<GARBAGE>
...

   118> 
   118> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62
   118> Flash config restore *FAILED*
   119> CONN led=0
   121> Wifi init, mode=AP
   123> Wifi uses DHCP, hostname=esp-link
   127> "ip": "0.0.0.0"
   129> "netmask": "0.0.0.0"
   132> "gateway": "0.0.0.0"
   134> "hostname": "<null>"
   137> sleep enable,type: 2
   139> Httpd init, conn=0x3fff2a6c
   143> Serbridge pins: reset=12 isp=13 swap=0
   155> Reset cause: 0=normal
   156> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   156> Flash map 1MB:512/512, manuf 0xE0 chip 0x4014
   160> ** esp-link v2.2.3 - 2016-06-21 21:58:48 - 1bcdc62: ready, heap=25112
   167> initializing user application
   170> Waiting for work to do...
   173> mode : softAP(5e:cf:7f:0b:ce:46)
   177> add if1
   179> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   187> bcn 100
```

codehero, 2020-04-27:
Tried another fw version:

```
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 8Mbit(512KB+512KB)
jump to run user1 @ 1000

����n��r��n|�
               l�
                 l`b�
                         �|r�l�n�
                                   �n�
                                      l`��r�l�l�

                                                  l`��r�l�l�

                                                              l`��r�l
                                                                      �

                                                                       ll`rlrl����b
                                                                                    ��bp|

                                                                                         bp����b
                                                                                                ��nn

                                                                                                    Serbridge pins: reset=12 isp=13 swap=0

   119> 
   119> ** esp-link v3.2.47.alpha.22-g6895442-dirty-Lucjan
   120> Flash config restore *FAILED*
   120> CONN led=0
   122> SER led=14
   124> Wifi init, mode=AP
   126> Wifi uses DHCP, hostname=esp-link
   130> "ip": "0.0.0.0"
   132> "netmask": "0.0.0.0"
   135> "gateway": "0.0.0.0"
   137> "hostname": "<null>"
   140> sleep enable,type: 2
   143> Httpd init, conn=0x3fff424c
   146> No user file system found!
   149> Reset cause: 0=normal
   152> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   158> Flash map 1MB:512/512, manuf 0xE0 chip 0x4014
   162> ** esp-link v3.2.47.alpha.22-g6895442-dirty-Lucjan: ready, heap=18640
   169> SNTP timesource set to us.pool.ntp.org with offset 0
   174> initializing user application
   178> Waiting for work to do...
   181> mode : softAP(5e:cf:7f:0b:ce:46)
   185> add if1
   186> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   194> bcn 100
```

uzi18, 2020-04-27:
your flash is 8Mbit (1MB) where did you flashed initial/default settings?

codehero, 2020-04-27:
Yes, I used the following command:
`esptool --port /dev/ttyUSB0 --baud 115200 write_flash -fs 8m -ff 40m 0x00000 boot_v1.7.bin 0x1000 user1.bin 0xFC000 esp_init_data_default.bin 0xFE000 blank.bin`

codehero, 2020-04-27:
Also flash_id output:

```
esptool.py v2.5.1
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: XX:XX:XX:XX:XX:XX
Enabling default SPI flash mode...
Manufacturer: e0
Device: 4014
Detected flash size: 1MB
Hard resetting via RTS pin...
```

uzi18, 2020-04-27:
try to add argument `-fm dio` just after write_flash, you can try also with boot_1.6

boutermw, 2020-09-07:
Did you solve this, I have the exact same problem with an ESP-01.
I think I can't rely on flashsize from esptool output because that info is overwriten if not mistaking.

Technic1975, 2020-12-19:
Did anyone find the solution to this, I have the exact same problem and it is driving me crazy!

Technic1975, 2020-12-30:
Well I have a further development, although the AP never shows, I have discovered that if I erase the module, then flash the hello server sketch with Arduino 1.8.14 with my SSID and password details included, then flash ESP link over the top, minus the Blank.bin file, I am able to connect to ESP link on my pc over my local network with IP 192.168.0.65 and configure as I like.

I updated the firmware to NMEA link and am able to send GPS data as I had planned, the unit still won't set up an AP though if it cant connect to the specified network.

This is really frustrating as I feel I am very close, does anyone have any Ideas?

uzi18, 2020-12-30:
Have complete flash dump available if anyone is interested from working module

Technic1975, 2020-12-31:
> Have complete flash dump available from working module if anyone is interested

I would very much appreciate it if you could supply this.

uzi18, 2020-12-31:
here it is: https://github.com/uzi18/sterownik/tree/master/arduino/esp-link
next use my update from here: https://github.com/jeelabs/esp-link/issues/480#issuecomment-577457913

Technic1975, 2020-12-31:
> here it is: https://github.com/uzi18/sterownik/tree/master/arduino/esp-link
> next use my update from here: [#480 (comment)](https://github.com/jeelabs/esp-link/issues/480#issuecomment-577457913)

Thanks very much, lets see how I get on!

uzi18, 2020-12-31:
First go with full flash erase, next decompress image and flash it with DOUT to 0x0 address
if it is working ok you can update with user1 and user2 files from second link

Technic1975, 2020-12-31:
Thanks for that, probably a dumb question, but what do you use to decompress the image file?

Technic1975, 2020-12-31:
> Thanks for that, probably a dumb question, but what do you use to decompress the image file?

Was a dumb question lol, you meant unzip the rar then flash the image, just done it now lets see what happens!

Technic1975, 2020-12-31:
That doesn't work I am afraid, It is stuck in a boot loop, the AP shows up for a second and I can connect to it then it drops out. This is what has happened with all other attempts to flash this.

Technic1975, 2020-12-31:
> First go with full flash erase, next decompress image and flash it with DOUT to 0x0 address
> if it is working ok you can update with user1 and user2 files from second link

Well I don't know why, but after many attempts at flashing the image with various different settings in the ESP8266 download tool, with just a constant boot loop, I tried with the SPI autoset check box ticked, and hey presto! I am able to connect to the Lucjan AP and configure via 192.168.4.1, I also updated the FW to the latest version as suggested and this seems to have worked also.

I will keep playing with it to try and figure out what the problem is exactly as the settings it autoset to I had already tried.

Many thanks for your help with this, much appreciated, it's been driving me crazy!

uzi18, 2021-01-02:
It because of bad quality of spi flashes (DOUT is always supported fallback).
If you used to high frequency it will not boot correctly.

Technic1975, 2021-01-02:
> It's because of bad quality of spi flashes (DOUT is always supported fallback).
> Also if you used to high frequency it will not boot correctly, settings are created in flash durring first boot or after update.

Thanks for that, I that would make sense, I had it working but after a while became unstable, I tried to erase and re-flash with exactly the same settings and am stuck back in a boot loop, very frustrating!
It could be that this board is faulty, I am going to order another and try again.
Thanks for all your help and advice.

uzi18, 2021-01-02:
You can try to change only flash chip or send me your unit so will check on my own.

Technic1975, 2021-01-10:
Well I purchased a second module, same results! after a lot of frustration I have found that if I flash these modules very slowly at 9600 Baud rate that they work just fine, very unstable flash chips! thanks for all your help uzi18.

cquijano, 2021-01-29:
Had the same issue with my ESP8266X, 4MB. After a lot of tries and error I found the that the firmware [V1.0.4](https://github.com/jeelabs/esp-link/releases/tag/v1.0.4) works fine for me, any other ,more recent version, have this problem.
Maybe it is because the build process of this image has a specific image for 4MB.

uzi18, 2021-01-29:
You can try to change flash chip to 32Mb

uzi18, 2021-01-29:
What is your flash chip ID?

cquijano, 2021-01-29:
@uzi18 

```
carlos@Sirio:~$ esptool.py chip_id
esptool.py v3.0
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Chip ID: 0x0063dfea
Hard resetting via RTS pin...

carlos@Sirio:~$ esptool.py flash_id
esptool.py v3.0
Found 2 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Manufacturer: 68
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```