jeelabs/esp-link #491: Wiring of  "GPIO2: optionally connect green LED to 3.3V (indicates wifi status)"

tonertiffi, 2020-04-29:
HI, i like the idea off the led to indicate status,but to get my head around the wiring.

Using a ESP-01

I assume the statement means the following GPIO2>>>>>green led (-) leg>>>>(+)leg to 3.3v.
No resistor
Can anyone clarify that.

mr-sneezy, 2020-04-29:
Add a resistor or else you rely on the ESP-01's own current limiting on the GPIO2 pin, if it has any. Anything from 250ohm to 500ohm betweeen the 3.3V pin +leg will be fine.

tonertiffi, 2020-04-29:
Hi Thank you for the reply.

Could I wire it up as per diagram?

[cid:image003.jpg@01D61E0B.DF822450]

tonertiffi, 2020-04-29:
Hi

Attached drawing

I assume that’s what you referring to.
Been allot off work to design the circuit, also very expensive to print it in South Africa, around 120$ so it is very important that I get it right 1st time.

Thank you for your help, much appreciated.

[cid:image001.jpg@01D61E0D.6ACF31A0]

tonertiffi, 2020-04-29:
[image]

mr-sneezy, 2020-04-29:
The LED part of the circuit is fine above.