jeelabs/esp-link #492: Trying to get OpenTherm to work with ESP link

Lars-, 2020-05-17:
Hi,

I'm trying to setup an OpenTherm Gateway with ESP link. So far, everything seems connected and working. The weird thing is that after 18 messages, nothing will be sent to port 23, the webbrowser or anything for that matter.

When I close the (telnet) connection and restart it, I will get 18 new messages. 

What am I doing wrong and how do I fix this?

I am using v2.2.3 by the way.

Would be great to hear from someone soon.

uzi18, 2020-05-17:
try this one:
[esp-link-v3.2.47.alpha.22-g6895442-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/4100794/esp-link-v3.2.47.alpha.22-g6895442-dirty-Lucjan.tgz.zip)

Lars-, 2020-05-18:
No improvements... Now it stops after 16 messages