jeelabs/esp-link #493: How can I reboot ?

povlhp, 2020-06-10:
I have mine runningon a Wemos board connected to my Visonic alarm system. Once in a while, I might lose comms on the serial port, I am still able to access the webserver.

Powercycling the ESP-Link has fixed the issues every single time. So how do we do a real reboot of the device ?  Not sure what have gone wrong, maybe out-of-sync on serial port, or something else ?

tve, 2020-06-11:
This type of question is best suited for the https://forum.micropython.com/
To reboot, have you tried the machine module?

tve, 2020-06-11:
Whoops, you can rebvoot using the UI and you can send it the equivalent rest request.
Does the log in the UI show an error?