jeelabs/esp-link #494: [REQUEST] esp-link firmware on Sonoff ZBBridge (ESP8266-based Zigbee Bridge from ITead)

Hedda, 2020-06-12:
Please consider porting esp-link to Sonoff ZBBridge (new Sonoff branded Zigbee Bridge by ITead).

- https://www.itead.cc/sonoff-zbbridge.html
  - https://sonoff.tech/product/smart-home-security/zbbridge

Then please try to enable serial bridge communication for the Zigbee MCU that is also inside this.

FYI, Tasmota and Zigbee-herdsman / Zigbee2MQTT developers have already started hacking it, see:
- https://github.com/arendst/Tasmota/issues/8583
- https://github.com/Koenkk/zigbee-herdsman/issues/168

Background and more information:

ITead recently started selling this Sonoff ZBBridge as an inexpensive Sonoff Zigbee Bridge that looks to be based on a similar concept as their previous Sonoff RF Bridge which there already are several alternative ESP8266 firmware available. 

It is based on an ESP8266 ("ESP8266EX") and more interestingly also contain a very powerful Zigbee 3.0 capable radio MCU from Silicon Labs EFR32 (Mighty Gecko) series and its off-the-shelf purpose is to be a Zigbee 3.0 to WiFi bridge/gateway/controller/hub for the new Sonoff branded Zigbee devices.

- https://www.cnx-software.com/2020/04/16/sonoff-zbbridge-wifi-to-zigbee-gateway/

This new Sonoff ZBBridge could be the perfect network-attached hardware for a local Zigbee to WiFi bridge/gateway if it was not for the case that it needs to be hacked to flash other firmware on its built-in ESP8266, ESP8266 firmware like esp-link (and other firmware for ESP8266).

The idea would be to first run ESP8266 firmware like esp-link on Sonoff ZBBridge and use the serial bridge function to give access to the Silicon Labs EFR32 Zigbee MCU that is also inside that device.

After that one would via the serial bridge function flash a new firmware on the Silicon Labs EFR32 Zigbee MCU, like Silicon Lavs standard Zigbee coordinator firmware as it is (as hopefully that already have the correct standard firmware that will be needed and then a developer with access to the official Silicon Labs SDK och Zigbee stack also needs to build a firmware for correct pin layout. 

From the teardown so far  it even looks like Sonoff are reusing the same injection moulds as for the Sonoff RF Bridge housing/enclosure which Tasmota already supports

- https://sonoff.tech/product/accessories/433-rf-bridge
- https://www.itead.cc/sonoff-rf-bridge-433.html
- https://www.itead.cc/wiki/Sonoff_RF_Bridge_433

A few popular open-source home automation projects that could possibly take advantage of the Zigbee MCU inside this device if had they had serial bridge access to the Zigbee MCU are Home Assistant's ZHA integration component (via  zigpy and bellows) as well as Zigbee2MQTT and IoBroger (both via zigbee-herdsman):

- https://www.home-assistant.io/integrations/zha/
- https://github.com/zigpy/bellows  
- https://github.com/zigpy/
- https://github.com/Koenkk/zigbee-herdsman
- https://www.zigbee2mqtt.io/

tve, 2020-06-12:
Thanks for your interest in esp-link. I'm afraid at this point esp-link is in maintenance mode and I'm not planning to add new features and what you're proposing would be a major amount of work...

Hedda, 2021-03-11:
FYI, tip now is to instead use Tube's Zigbee Gateways open-source hardware by @tube0013 (which is tested with ESPHome FW).

Based on WT32-ETH01 (Wireless-Tag) ESP32 board which has wired Ethernet that makes for a much more stable serial connection.

He designed two variants; one has a Silicon Labs EFR32 Series 2 module and one that has a Texas Instruments CC2652P module.

https://github.com/tube0013/tube_gateways

[https://github.com/zigpy/zigpy/discussions/584](https://github.com/zigpy/zigpy/discussions/584)

https://www.tubeszb.com/shop/coordinators/2

[image]

[image]

Hedda, 2021-09-06:
FYI, thegroove made a simple to use Serial-to-IP bridge appliance firmware for Sonoff Zigbee Bridge running ESPHome firmware:

https://github.com/thegroove/esphome-zbbridge

oxan and thegroove also look to have made two alternative serial server code component solutions for ESPHome firmware:

https://github.com/oxan/esphome-stream-server

https://github.com/thegroove/esphome-serial-server

thegroove  even made an experimental version with Zeroconf support for automatic network discovery by ZHA in Home Assistant:

https://github.com/thegroove/esphome-zha-ezsp-zeroconf

https://www.home-assistant.io/integrations/zha/#discovery-via-usb-or-zeroconf

Note that Zigbee2MQTT is another popular alternative to Home Assistant's ZHA for use with this type of Serial-to-IP bridge:

https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html