jeelabs/esp-link #496: Is 8-N-1 right for this?

elvisthedj, 2020-08-29:
[image]

Trying to breathe some new life into an old amp that only supports control via RS232. 

I'm set to 4800 8N1, but the manual doesn't mention anything about the stop bit.. just 4800, 8, no parity and no handshake (no CR or LF either).

I should get an OK when I send AT, but nothing I send gets a reply.

I have not tested this with any other method because i don't yet have what I need (usb -> db9 rs232 on the way), but I just wanted to see if someone had an option for me to try.

I used my scope to verify the signaling is going to the right pin, but I get no response. I'm probably going to have to wait until my cable shows up, but thought maybe there was a slight chance that someone else had gone through something similar and came up with an answer.

Thanks!

uzi18, 2020-08-29:
@elvisthedj any type/model of this amp?

elvisthedj, 2020-08-29:
[Matrix Audio MRC-84e](http://www.drivehq.com/file/df.aspx/shareID3748302/fileID532235526/)

I just realized that even with Vin, I'm not getting the voltage necessary for RS232. I'm going to try it with an external supply for the ttl-rs232 chip and see if I get a proper signal.

uzi18, 2020-08-29:
It is hard to find anything more, you need 3v3 to rs232 converter and null modem cable

elvisthedj, 2020-08-29:
I'm getting a solid RS232 signal on my scope, but still no reply from device. I notice that the url in the esp log includes h=xxxxx ... Does this mean I'm sending hex values? I'm thinking I need to just send ASCII.

uzi18, 2020-11-26:
please attach some logs