jeelabs/esp-link #498: Failed to reset the board, upload failed

kokojza, 2020-09-17:
Hi guys!
I have a problem with the OTA upload. I followed the instructions and after the setup it was working like heaven, but after a while I got an error:

`Arduino: 1.8.10 (Windows 10), Board: "Arduino/Genuino Uno"

Sketch uses 25226 bytes (78%) of program storage space. Maximum is 32256 bytes.
Global variables use 1423 bytes (69%) of dynamic memory, leaving 625 bytes for local variables. Maximum is 2048 bytes.
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/arduinoOTA -address 192.168.1.108 -port {upload.network.port} -sketch C:\Users\user\AppData\Local\Temp\arduino_build_417960/main_v10.3.ino.hex -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return} 
Resetting the board
Failed to reset the board, upload failed
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.`

No change in wiring, no change in op system (maybe an update? I don't remember).

drunkscientist, 2020-09-28:
getting the same error on a nano, which i only tried because absoloutely nothing worked on the mega1284p i wanted to use...

uzi18, 2020-09-28:
Maybe  they should be added to programmer function
What version of esp-link? 
What avr cpu signature?
Did you tried to check debug logs?

drunkscientist, 2020-09-30:
> Maybe they should be added to programmer function
like i should do that or thats a long term solution? i dont really know what that means
> What version of esp-link?
2.2.3 or whatever 'newest' stable release is.
> What avr cpu signature?
i dont know what that means. the arduino nano has an atmega 328.
> Did you tried to check debug logs?
the point of using the arduino IDE is it handles all the minuta and spits out anything and everything in a consistent place. currently that place holds about as much as
```
Resetting the board
Failed to reset the board, upload failed
An error occurred while uploading the sketch
```
do logs get generated somewhere else?