jeelabs/esp-link #499: Wifi to Serial and vice versa

vinura, 2020-09-19:
Is it possible to do the following?

ESP[Serial to Wifi]--------------> Wifi Bridge -----------> ESP[Wifi to Serial]

I am trying to connect Pixhwak to a Ground Station using the above method. Is it possible to broadcast serial to wifi from one ESP and receive it from another ESP and direct it to serial connection again?

mr-sneezy, 2020-09-19:
Slightly off your actual question, I see you mention MAVlink. I've done it with this below from Pixhawk to UDP into Mission Planner.
https://ardupilot.org/copter/docs/common-esp8266-telemetry.html
I would think it would work through a WiFi repeater also.
If you really need serial via WiFi at the GCS and not UDP then you may have to experiment.

vinura, 2020-09-20:
@mr-sneezy Yes i got it to working with Mission Planner without any problem using UDP. But no I have this GCS Unit Called Tengri [http://tengriuav.com/]. It supports Mavlink over serial, RFD, and USB. I have one pair of RFD 900x. But I also have 4 drones. It is a pain to swap RFDs all the time. Since I want to ditch the Mission Planner altogether. I am looking to have that wifi to serial connection using esp-link. 

Simply put is esp link takes Wifi input and give to serial. rather than other way around

tapi35, 2020-09-30:
Vinura, I want to create the same diagram for an other application : canceling wire between an electronic instrument and the sound module.  If you find how to make this, it will be nice to share, I will do the same.
Thanks

vinura, 2020-10-01:
@tapi35 I am looking at ESP Now Protocol and how that I could use it for this purpose. Will share an update. So far I found one solution but it really didn't work. Please take a look. http://cs107e.github.io/guides/extras/uart-wifi-bridge/

uzi18, 2020-11-26:
we should implement it into esp-link

vinura, 2021-03-12:
@tapi35   I was able to use socat on raspberry pi and convert udp port back to serial.
socat -d -d UDP-LISTEN:14550 /dev/ttyS0,b921600,raw,echo=0

but this is costly for just this purpose.

rkubera, 2021-04-25:
look at: https://github.com/rkubera/ESP8266-RC-Transmitter-Receiver I was tested this with pixhawk.