jeelabs/esp-link #504: cant reset board

drunkscientist, 2020-11-25:
have tried with an atmega1284p, have tried with an uno and a nano, both mega328, and a mega2560, none can flash a sketch from arduino, ```
 Resetting the board
Failed to reset the board, upload failed
An error occurred while uploading the sketch

```

serial works fine, reset from web console works fine, but cannot flash sketch and im stumped

Meizel, 2020-12-12:
Hi drunkscientist,

Try to load v0.9.2.2 AT Firmware.bin with 0x00000

This works for me with the ESP-01s after that I just reload it with esp-link

[ESP8266 Flasher (resetten).zip](https://github.com/jeelabs/esp-link/files/5682428/ESP8266.Flasher.resetten.zip)

imbalaji16, 2021-01-22:
we are using esp 8266 for serial data monitoring. recently esp getting reset frequently and when i try to connect the esp signal on my laptop it says unable to connect pls suggest us
[image]

uzi18, 2021-01-22:
First try to stabilise voltage with additional 1000uF capacitor on 3V3 line near esp
And you should open new issue for this problem

uzi18, 2021-01-22:
@drunkscientist how did you choose board ip?

drunkscientist, 2021-01-23:
[image]

@uzi18 i picked the ip by clicking on it from the dropdown menu. the entirety of my log (with verbose everything turned on) is 

`Arduino: 1.8.13 (Linux), Board: "Arduino Nano, ATmega328P"

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/robert/.arduino15/packages -hardware /home/robert/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/robert/.arduino15/packages -libraries /home/robert/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10813 -build-path /tmp/arduino_build_847676 -warnings=none -build-cache /tmp/arduino_cache_244521 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/robert/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/robert/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /tmp/arduino_modified_sketch_103250/Blink.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/robert/.arduino15/packages -hardware /home/robert/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/robert/.arduino15/packages -libraries /home/robert/Arduino/libraries -fqbn=arduino:avr:nano:cpu=atmega328 -ide-version=10813 -build-path /tmp/arduino_build_847676 -warnings=none -build-cache /tmp/arduino_cache_244521 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.arduinoOTA.path=/home/robert/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=/home/robert/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0 -prefs=runtime.tools.avrdude.path=/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/home/robert/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17 -verbose /tmp/arduino_modified_sketch_103250/Blink.ino
Using board 'nano' from platform in folder: /home/robert/.arduino15/packages/arduino/hardware/avr/1.8.3
Using core 'arduino' from platform in folder: /home/robert/.arduino15/packages/arduino/hardware/avr/1.8.3
Detecting libraries used...
/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/robert/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/robert/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs /tmp/arduino_build_847676/sketch/Blink.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/robert/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/robert/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs /tmp/arduino_build_847676/sketch/Blink.ino.cpp -o /tmp/arduino_build_847676/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
/usr/bin/arduino-ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_847676/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR -I/home/robert/.arduino15/packages/arduino/hardware/avr/1.8.3/cores/arduino -I/home/robert/.arduino15/packages/arduino/hardware/avr/1.8.3/variants/eightanaloginputs /tmp/arduino_build_847676/sketch/Blink.ino.cpp -o /tmp/arduino_build_847676/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /tmp/arduino_cache_244521/core/core_arduino_avr_nano_cpu_atmega328_c175a3e84e05059895ae32adccdce175.a
Linking everything together...
/home/robert/.arduino15/packages/arduino/tools/avr-gcc/7.3.0-atmel3.6.1-arduino7/bin/avr-gcc -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o /tmp/arduino_build_847676/Blink.ino.elf /tmp/arduino_build_847676/sketch/Blink.ino.cpp.o /tmp/arduino_build_847676/../arduino_cache_244521/core/core_arduino_avr_n [... cut ...]
Global variables use 194 bytes (9%) of dynamic memory, leaving 1854 bytes for local variables. Maximum is 2048 bytes.
/home/robert/.arduino15/packages/arduino/tools/arduinoOTA/1.3.0/bin/arduinoOTA -address 192.168.1.203 -port {upload.network.port} -sketch /tmp/arduino_build_847676/Blink.ino.hex -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return} 
Resetting the board
An error occurred while uploading the sketch
Failed to reset the board, upload failed
`

uzi18, 2021-01-24:
@drunkscientist what is you esp-link version?

uzi18, 2021-01-24:
hints for uno from here https://www.gitmemory.com/issue/jeelabs/esp-link/509/743165574
add those 5 lines to the boards.txt: 
```
uno.upload.network.endpoint_upload=/pgm/upload
uno.upload.network.endpoint_sync=/pgm/sync
uno.upload.network.sync_return=204:SYNC 
uno.upload.network.endpoint_reset=/log/reset
uno.upload.network.port=80
```

drunkscientist, 2021-01-24:
so the 2 main issues there are that i have ditched arduino ide in favor of vs code/platformIO, and that the main goal is to flash a mega2560, which is apparently dubious

Albert245, 2022-11-03:
Hi, I've the same issue. Any update yet ??

uzi18, 2022-11-03:
@Albert245 what is your board selected by Arduino IDE ?

Albert245, 2022-11-03:
It's Uno. I've add lines into boards.txt but still not worked.

uzi18, 2022-11-03:
@Albert245 please paste (enable verbose programmer in options) log

Albert245, 2022-11-03:
Arduino: 1.8.13 (Windows 10), Board: "Arduino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\nhtc1\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\nhtc1\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\nhtc1\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10813 -build-path C:\Users\nhtc1\AppData\Local\Temp\arduino_build_323361 -warnings=none -build-cache C:\Users\nhtc1\AppData\Local\Temp\arduino_cache_535662 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose G:\University Project\test\Blink\Blink.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\nhtc1\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\nhtc1\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\nhtc1\OneDrive\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10813 -build-path C:\Users\nhtc1\AppData\Local\Temp\arduino_build_323361 -warnings=none -build-cache C:\Users\nhtc1\AppData\Local\Temp\arduino_cache_535662 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avrdude.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avr-gcc.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino7 -verbose G:\University Project\test\Blink\Blink.ino

Using board 'uno' from platform in folder: C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Using core 'arduino' from platform in folder: C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Detecting libraries used...

"C:\\Users\\nhtc1\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\nhtc1\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\nhtc1\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "C:\\Users\\nhtc1\\AppData\\Local\\Temp\\arduino_build_323361\\sketch\\Blink.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Generating function prototypes...

"C:\\Users\\nhtc1\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Users\\nhtc1\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\nhtc1\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\standard" "C:\\Users\\nhtc1\\AppData\\Local\\Temp\\arduino_build_323361\\sketch\\Blink.ino.cpp" -o "C:\\Users\\nhtc1\\AppData\\Local\\Temp\\arduino_build_323361\\preproc\\ctags_target_for_gcc_minus_e.cpp" -DARDUINO_LIB_DISCOVERY_PHASE

"C:\\Program Files (x86)\\Arduino\\tools-builder\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\nhtc1\\AppData\\Local\\Temp\\arduino_build_323361\\preproc\\ctags_target_for_gcc_minus_e.cpp"

Compiling sketch...

"C:\\Users\\nhtc1\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p [... cut ...]
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.

C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA -address 192.168.1.103 -port {upload.network.port} -sketch C:\Users\nhtc1\AppData\Local\Temp\arduino_build_323361/Blink.ino.hex -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return} 

Resetting the board
An error occurred while uploading the sketch

Failed to reset the board, upload failed

uzi18, 2022-11-03:
> C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0/bin/arduinoOTA -address 192.168.1.103 -port {upload.network.port} -sketch [...]

as You can see these lines are not resolved - wrong file modified?

uzi18, 2022-11-03:
ok I see it :
```
nano.upload.network.endpoint_upload=/pgm/upload
nano.upload.network.endpoint_sync=/pgm/sync
nano.upload.network.sync_return=204:SYNC 
nano.upload.network.endpoint_reset=/log/reset
nano.upload.network.port=80
```

please change nano to uno

Albert245, 2022-11-03:
##############################################################

uno.name=Arduino Uno

uno.vid.0=0x2341
uno.pid.0=0x0043
uno.vid.1=0x2341
uno.pid.1=0x0001
uno.vid.2=0x2A03
uno.pid.2=0x0043
uno.vid.3=0x2341
uno.pid.3=0x0243

uno.upload.tool=avrdude
uno.upload.protocol=arduino
uno.upload.maximum_size=32256
uno.upload.maximum_data_size=2048
uno.upload.speed=115200

uno.bootloader.tool=avrdude
uno.bootloader.low_fuses=0xFF
uno.bootloader.high_fuses=0xDE
uno.bootloader.extended_fuses=0xFD
uno.bootloader.unlock_bits=0x3F
uno.bootloader.lock_bits=0x0F
uno.bootloader.file=optiboot/optiboot_atmega328.hex

uno.build.mcu=atmega328p
uno.build.f_cpu=16000000L
uno.build.board=AVR_UNO
uno.build.core=arduino
uno.build.variant=standard

uno.upload.network.endpoint_upload=/pgm/upload
uno.upload.network.endpoint_sync=/pgm/sync
uno.upload.network.sync_return=204:SYNC
uno.upload.network.endpoint_reset=/log/reset
uno.upload.network.port=80

##############################################################

diecimila.name=Arduino Duemilanove or Diecimila

diecimila.upload.tool=avrdude
diecimila.upload.protocol=arduino

diecimila.bootloader.tool=avrdude
diecimila.bootloader.low_fuses=0xFF
diecimila.bootloader.unlock_bits=0x3F
diecimila.bootloader.lock_bits=0x0F

diecimila.build.f_cpu=16000000L
diecimila.build.board=AVR_DUEMILANOVE
diecimila.build.core=arduino
diecimila.build.variant=standard

## Arduino Duemilanove or Diecimila w/ ATmega328P
## ----------------------------------------------
diecimila.menu.cpu.atmega328=ATmega328P

diecimila.menu.cpu.atmega328.upload.maximum_size=30720
diecimila.menu.cpu.atmega328.upload.maximum_data_size=2048
diecimila.menu.cpu.atmega328.upload.speed=57600

diecimila.menu.cpu.atmega328.bootloader.high_fuses=0xDA
diecimila.menu.cpu.atmega328.bootloader.extended_fuses=0xFD
diecimila.menu.cpu.atmega328.bootloader.file=atmega/ATmegaBOOT_168_atmega328.hex

diecimila.menu.cpu.atmega328.build.mcu=atmega328p

## Arduino Duemilanove or Diecimila w/ ATmega168
## ---------------------------------------------
diecimila.menu.cpu.atmega168=ATmega168

diecimila.menu.cpu.atmega168.upload.maximum_size=14336
diecimila.menu.cpu.atmega168.upload.maximum_data_size=1024
diecimila.menu.cpu.atmega168.upload.speed=19200

diecimila.menu.cpu.atmega168.bootloader.high_fuses=0xdd
diecimila.menu.cpu.atmega168.bootloader.extended_fuses=0xF8
diecimila.menu.cpu.atmega168.bootloader.file=atmega/ATmegaBOOT_168_diecimila.hex

diecimila.menu.cpu.atmega168.build.mcu=atmega168

##############################################################

nano.name=Arduino Nano

nano.upload.tool=avrdude
nano.upload.protocol=arduino

nano.bootloader.tool=avrdude
nano.bootloader.unlock_bits=0x3F
nano.bootloader.lock_bits=0x0F

nano.build.f_cpu=16000000L
nano.build.board=AVR_NANO
nano.build.core=arduino
nano.build.variant=eightanaloginputs

## Arduino Nano w/ ATmega328P
## --------------------------
nano.menu.cpu.atmega328=ATmega328P

nano.menu.cpu.atmega328.upload.maximum_size=30720
nano.menu.cpu.atmega328.upload.maximum_data_size=2048
nano.menu.cpu.atmega328.upload.speed=115200

nano.menu.cpu.atmega328.bootloader.low_fuses=0xFF
nano.menu.cpu.atmega328.bootloader.high_fuses=0xDA
nano.menu.cpu.atmega328.bootloader.extended_fuses=0xFD
nano.menu.cpu.atmega328.bootloader.file=optiboot/optiboot_atmega328.hex

nano.menu.cpu.atmega328.build.mcu=atmega328p

## Arduino Nano w/ ATmega328P (old bootloader)
## --------------------------
nano.menu.cpu.atmega328old=ATmega328P (Old Bootloader)

nano.menu.cpu.atmega328old.upload.maximum_size=30720
nano.menu.cpu.atmega328old.upload.maximum_data_size=2048
nano.menu.cpu.atmega328old.upload.speed=57600

nano.menu.cpu.atmega328old.bootloader.low_fuses=0xFF
nano.menu.cpu.atmega328old.bootloader.high_fuses=0xDA
nano.menu.cpu.atmega328old.bootloader.extended_fuses=0xFD
nano.menu.cpu.atmega328old.bootloader.file=atmega/ATmegaBOOT_168_atmega328.hex

nano.menu.cpu.atmega328old.build.mcu=atmega328p

## Arduino Nano w/ ATmega168
## -------------------------
nano.menu.cpu.atmega168=ATmega168

nano.menu.cpu.atmega168.upload.maximum_size=14336
nano.menu.cpu.atmega168.upload.maximum_data_size=1024
nano.menu.cpu.atmega168.upload.speed=19200

nano.menu.cpu.atmega168.bootloader.low_fuses=0xff
nano.menu.cpu.atmega168.bootloader.high_fuses=0xdd
nano.menu.cpu.atmega168.bootloader.extended_fuses=0xF8
nano.menu.cpu.atmega168.bootloader.file=atmega/ATmegaBOOT_168_diecimila.hex

nano.menu.cpu.atmega168.build.mcu=atmega168

> I've changed at the first place :'(

uzi18, 2022-11-03:
@Albert245 so arduino ide does not take it into account, wrong file? try programmers.txt

Albert245, 2022-11-03:
Still that error :'(

uzi18, 2022-11-03:
@Albert245 you should check these paths:
```
Using board 'uno' from platform in folder: C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Using core 'arduino' from platform in folder: C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
```

Albert245, 2022-11-04:
I've tried. But it not start at the first time. But after I fixed at both 'C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6' and 'C:\Program Files (x86)\Arduino\hardware\arduino\avr' in 3 txt (boards, platform, programmers) it finally work.
Thanks for supporting me @uzi18 sir

Albert245, 2022-11-04:
But can I ask why port 80 but not 23 or 2323 ? I though the only way to access the reset pin is in port 2323.

uzi18, 2022-11-04:
@Albert245 these are different APIs
ports 23 and 2323 are telnet or tcp/serial bridge - this some users use with avrdude programmer directly
ports 80 with /pgm/sync and /pgm/upload are ArduinoOTA API

uzi18, 2022-11-04:
@Albert245 could you please find what file does the job for you? It will be answare for future issues like this one.

Albert245, 2022-11-04:
It's boards.txt at this dir 'C:\Users\nhtc1\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6'.

Albert245, 2022-11-04:
And I want to use esp-link on the Arduino-CLI, it went wrong at port 23, 2323, 80. What port should I use for this ? 
The error for port 23 or 2323 and the current sketch on Uno was serial at 115200:

>  G:\University Project\test>arduino-cli upload -p net:192.168.1.103:23 --fqbn arduino:avr:uno SerialSample
> avrdude: ser_drain(): read error: The parameter is incorrect.
> 
> avrdude: ser_drain(): read error: The parameter is incorrect.
> 
> avrdude: ser_drain(): read error: The parameter is incorrect.
> 
> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x3c
> avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x41
> avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x72
> avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x64
> avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x75
> avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x69
> avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x6e
> avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x6f
> avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x20
> avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x55
> avrdude: ser_drain(): read error: The parameter is incorrect.
> 
> Error during Upload: Failed uploading: uploading error: exit status 1

And if I used port 80 or 23 & 2323 when the current sketch on board is not open serial. The resp is 1 kind of hex for 10 times:

> avrdude: stk500_recv(): programmer is not responding
> avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xc2
> avrdude: stk500_recv(): programmer is not responding
> avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xc2
> avrdude: stk500_recv(): programmer is not responding
> avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xc2
 Difference hex number at a time.

uzi18, 2022-11-04:
@Albert245 not sure how to run arduinoota via arduino-cli

uzi18, 2022-11-04:
@Albert245 you can try -P option to select programmer like network_cmd or arduino_ota

look:
https://github.com/arduino/ArduinoCore-avr/blob/42fa4a1ea1b1b11d1cc0a60298e529d37f9d14bd/platform.txt#L129-L136

paste here output of this commands:
```
arduino-cli board details -b arduino:avr:uno
arduino-cli board list --discovery-timeout 10s
```

uzi18, 2022-11-04:
```
Programmers:               Id                                                       Name
                           buspirate                                                BusPirate as ISP
                           arduinoasisp                                             Arduino as ISP
                           arduinoasispatmega32u4                                   Arduino as ISP (ATmega32U4)
                           parallel                                                 Parallel Programmer
                           usbGemma                                                 Arduino Gemma
                           avrisp                                                   AVR ISP
                           arduinoisp                                               ArduinoISP
                           arduinoisporg                                            ArduinoISP.org
                           stk500                                                   Atmel STK500 development board
                           avrispmkii                                               AVRISP mkII
                           usbtinyisp                                               USBtinyISP
                           jtag3                                                    Atmel JTAGICE3 (JTAG mode)
                           jtag3isp                                                 Atmel JTAGICE3 (ISP mode)
                           atmel_ice                                                Atmel-ICE (AVR)
                           usbasp                                                   USBasp

```

so think about declaration of own programmer based on arduino_ota lines

Albert245, 2022-11-06:
I will try

uzi18, 2022-11-07:
@Albert245 waiting for feedback