jeelabs/esp-link #505: Question: ESP-01 keeps freezing when using MQTT - what to do?

sixtyfive, 2020-11-29:
For remote uploading to an attached Arduino Nano as well as the dynamic-webpage-feature, ESP-Link has been rock-solid-stable for me, no single crash or freeze or anything, ever, now matter how many days it has to go without reboot.

Now I want to start using the MQTT feature, too, and, while beginning to work it into the Nano's sketch, thought I'd already enable MQTT in ESP-Link's admin interface. At first that worked just fine, including the built-in status messages. Unfortunately, after a while the ESP-01 just ... freezes, doesn't even respond to pings anymore, and has to be cold-rebooted. It seems to be working for longer with the status messages off, but that's just my feeling.

Is there anything I could do to get to the bottom of this?

Meizel, 2020-12-12:
Hi @sixtyfive

You can try this version.

[esp-link-v3.2.47.alpha.22-g6895442-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/5682439/esp-link-v3.2.47.alpha.22-g6895442-dirty-Lucjan.tgz.zip)

sixtyfive, 2020-12-12:
A pre-compiled binary, zipped after having been tgz'd, by someone not in the list of contributors of this project so far, who exclusively has forks in his account, uses a creepy-looking wolf as their avatar and otherwise gives no information about themselves, whatsoever? No, thank you.

@tve, you might want to take a look here.

uzi18, 2020-12-12:
@sixtyfive it is my own precompiled binary, because release is in tgz and github accepts only zip
https://github.com/jeelabs/esp-link/issues/480#issuecomment-577457913

sixtyfive, 2020-12-12:
Thank you for the clarification, @uzi18. Would you be able to tell what branch / with what patches / PRs / etc. applied that binary was built from?

uzi18, 2020-12-12:
@sixtyfive it is built from my fork (master branch) with latest fixes, but not sure if it contains query-ip command, will check later

sixtyfive, 2020-12-13:
I see! What does the query-ip command do? And, seeing as your master branch is 60 commits behind jeelabs:master, what are the latest fixes that you're referring to? I'll gladly build myself if a later version than 3.0.14 is likely to fix the freezing problem :-)