jeelabs/esp-link #508: Esp link as an serial to usb endpoint? Fully transparent serial bridge?

rlesac, 2020-12-06:
Hi!
Is there a way to use ESP-LINK as a receiver for another ESP-LINK?
So I can plug the ESP-LINK to an UART->USB and communicate through it to another ESP-LINK?
Without virtual serial drivers.

RoganDawes, 2021-02-12:
Yes, it should be possible, assuming:
- You can provide enough power to the ESP8266 via your USB connection. Most USB-UART adapters, while providing 3v3 lines, can only provide 50-100mA, which is not enough to power an ESP8266.
- You can modify the esp-link code to make an outbound TCP connection to the target esp-link instance (or alternatively, hack up your own Arduino or whatever code to do so). Depending on your requirements, you may need to have the code retry in a loop, or possibly set the CTS pins while the TCP connection is down, etc.

Conceptually, it's not an impossible task, but may require a bit of coding to get it working reliably. I'm assuming reliability and ease of use is a requirement, otherwise making use of virtual drivers would probably be an easier approach.

For bonus points, since the ESP8266 supports automatic baud rate detection on its serial port, you could have it automatically detect the baud rate from the connected USB-UART and pass those parameters to the target esp-link via an HTTP call.

However, if I was building this, I'd skip the USB-UART plus ESP8266 entirely, and use an ESP32-S2, which has a native USB interface. From there, implement USB-UART to TCP functionality, which, since both UAB-UART and TCP functionality already exists, is pretty much gluing the two pieces together, potentially with some configuration interface on top, accessed either via USB or WiFi/HTTP.

uzi18, 2021-02-12:
Yes there is demand for such feature, but this client mode should be added to. 
One need to find some spere time to develop it.

vinura, 2021-03-12:
yes there is a demand, I am searching for a proper solution since last year. The closest I cam is to use a raspberry pi and convert udp port back to serial using socat. 

socat -d -d UDP-LISTEN:14550 /dev/ttyS0,b921600,raw,echo=0

rraat, 2021-04-10:
Doesn't the, stock AT firmware (Hayes compatible modem) Work for the dial out side?

RoganDawes, 2021-04-12:
The AT firmware can make a dial-out connection, when instructed externally. The point of this discussion (as I understand it) is for that outbound connection to be made automatically, possibly on powerup, and then maintained going forward, should the connection drop for any reason.