jeelabs/esp-link #509: Flashing 2560 with esplink.

Capiten, 2020-12-11:
Hi

I now there are thousands of posts already asking the same (how to flash 2560 with esplink!?) but either their solution didn't worked for me or I didn't understand the solution. My main goal will be to be able to upload new codes trought internet to an at2560, outside of its network, but for know I'm just trying to upload within the same network.
I managed to upload to an arduino uno, with the esplink firmware inside an esp8266 and with the esplink service named as 'arduino' (making is appear as a network port on arduino IDE - the virtual com port with 'HW virtual serial port' did not work for me).
I also had to add those 5 lines to the boards.txt:
uno.upload.network.endpoint_upload=/pgm/upload
uno.upload.network.endpoint_sync=/pgm/sync
uno.upload.network.sync_return=204:SYNC
uno.upload.network.endpoint_reset=/log/reset
uno.upload.network.port=80

And it worked.
But at2560 doesn't go that easily, even with the same lines on boards.txt (started with mega instead of uno), it doesnt work. I flashed the STk500v2 bootloader as some mentioned but didn't changed anything. When I try to do the same precedure on at2560 in arduino IDE as I do for arduino uno, the log says this:

Resetting the board
Waiting for the upload to start
An error occurred while uploading the sketch
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
Error flashing the sketch:sync abandoned after 8 attempts

On esplink 'uc Console' I can see the uart0 prints the atmega2560 is doing and i can even reset the board trough esplink, so connections don't seem to be the problem.
I'm using Arduino IDE 1.18.13.
I'm open to other options, I even tried the 'esp avr programmer'  where you have to upload the .hex into and .htm webpage, but the project seemed to have been abandoned 4 years ago and the webpage appears with to many errors to allow the flashing.
However, esplink seems really cool, it's just a matter of learning how to use it properly.

Can anyone help me? Just point me the direction.

Thank you.
Cheers, Luís Pereira.

JohnOH, 2020-12-11:
I wonder if the boot loader (STk500v2) is a match for the protocol being triggered by Arduino as it tries to upload via your network port. Perhaps you could upload to the 2560 using a standard serial line and note down the verbose output of the avrdude upload process:

/Users/johare/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude -C/Users/johare/Library/Arduino15/packages/MegaCore/hardware/avr/2.0.3/avrdude.conf -v -patmega2560 -carduino -P/dev/cu.usbmodem14101 -b115200 -D -Uflash:w:/var/folders/6c/gxr4fz9s0_1cfcfnq7b8fmr40000gp/T/arduino_build_773057/RFxConsole.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/johare/Library/Arduino15/packages/MegaCore/hardware/avr/2.0.3/avrdude.conf"
         User configuration file is "/Users/johare/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem14101
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

Then check what protocol is used when connecting via the network port. Which core are you using to build the code, my values above are produced by the MegaCore.

Capiten, 2020-12-11:
Hi John, thank you for the quick repply. 
Here's the output with usb cable, where upload goes smoothly:
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf -v -patmega2560 -cwiring -PCOM10 -b115200 -D -Uflash:w:C:\Users\Luis\AppData\Local\Temp\arduino_build_835725/Blink_com_prints_para_fazer_debug_da_serial_e_led.ino.hex:i 

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM10
         Using Programmer              : wiring
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega2560
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

And here is the output where it goes bananas:
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\Luis\\AppData\\Local\\Temp\\arduino_build_835725/Blink_com_prints_para_fazer_debug_da_serial_e_led.ino.elf" "C:\\Users\\Luis\\AppData\\Local\\Temp\\arduino_build_835725/Blink_com_prints_para_fazer_debug_da_serial_e_led.ino.hex"
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\Luis\\AppData\\Local\\Temp\\arduino_build_835725/Blink_com_prints_para_fazer_debug_da_serial_e_led.ino.elf"
Sketch uses 2746 bytes (1%) of program storage space. Maximum is 253952 bytes.
Global variables use 206 bytes (2%) of dynamic memory, leaving 7986 bytes for local variables. Maximum is 8192 bytes.
C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/arduinoOTA -address 192.168.1.96 -port 80 -sketch C:\Users\Luis\AppData\Local\Temp\arduino_build_835725/Blink_com_prints_para_fazer_debug_da_serial_e_led.ino.hex -upload /pgm/upload -sync /pgm/sync -reset /log/reset -sync_exp 204:SYNC 
Resetting the board
Waiting for the upload to start
An error occurred while uploading the sketch
Connecting to board ...  done
Uploading sketch ...  done
Flashing sketch ...  done
Error flashing the sketch:sync abandoned after 8 attempts

About the core, not sure what to tell you, its's something I don't recall chaging, but there is this line before the output i'm sending you above: Using precompiled core: C:\Users\Luis\AppData\Local\Temp\arduino_cache_6357\core\core_arduino_avr_mega_cpu_atmega2560_0c812875ac70eb4a9b385d8fb077f54c.a

Not sure if thats the core you asked about, sorry

Thank you.
Cheers, Luís Pereira.

Capiten, 2020-12-16:
Can anyone help me? Or just give me a clue of what might be?
Thank you.

JohnOH, 2020-12-16:
Hi @Capiten, I'm sorry I could not add to the work you did at my request. The things I find odd are:
-address 192.168.1.96 -port 80 
I don't see port 80 referenced in the esplink documentation.

The esplink documentation talks about the inbuilt STK500V1 support, I understand that you have an STK500V2 bootloader in your 2560. Although I do think esplink passes through the data stream. With your settings Arduino is using -cwiring  as its protocol to interact with the 2560 bootloader. I wonder if that is correct. 

It feels like a bootloader issue to me.

Capiten, 2020-12-16:
Hi John
Thank you for the repply.
About the "-address 192.168.1.96 -port 80", thats probably because I'm using the "Network port" on arduino, that shows up when esplink service is named "arduino". The other way I saw, involving a virtual com port, did not work either, don't recall the error but if needed I can repeat that error.
The -cwiring shows when I upload the 328p, and that one uploads just fine trough esplink =\

I could try the STK500V1 and see how it goes.
Thank you.

uzi18, 2020-12-17:
For flashing mega board with 2560 you should use megaflash script, it uses different api to support stk500v2 bootloader. 
Do you need more details ?

Capiten, 2020-12-28:
Hi
Sorry for the long delay in my feedback but with christmass and all I didn't manage to get to this.
Anyway, I tried the megaflash.sh script and I get the error:
**Error resetting AVR**

The hardware connections don't seem to be the problem because I can reset trhe atmgea2560 trough esplink on uC Console.
The ip seems ok because thats the ip I'm using to see esplink webpage and I can ping that IP on the WSL on windows (i'm on win10 with the wsl).
Looking at the megaflash script, I can see the probem comes from this if condition:
**if [[ $? != 0 || "$sync" != 204 ]]; then
  echo "Error resetting AVR" >&2
  exit 1
fi**

I splitted the conditions being testes, like this:
**if [[ $? != 0 ]]; then
  echo "Error resetting AVR_$? != 0" >&2
  exit 1
fi
 if [[ "$sync" != 204 ]]; then
  echo "Error resetting AVR_sync" >&2
  echo "$sync"
  exit 1 
fi**

And i can see i get the Error resetting AVR_sync printed out, and an 404 after. So i'm getting an http 404 error, so the megaflash seems to fail to get in touch with the ip I give it...but as I said, the IP is correct, and I can ping it.
Any more ideas?
Thank you.

uzi18, 2020-12-28:
What are your boards and how did you connect to esp-link, in my opinion tx line is in collision with usb bridge if you use Arduino Mega, there is a trick we use for such case

Capiten, 2020-12-28:
Well the esp8266 is connected to the 2560 via 3.3v, GND, TX to RX and RX to TX, and GPIO0 to Reset on mega.
The mega2560 is connected with isb cable to the laptop...could that be it?

uzi18, 2020-12-28:
So there is a conflict between usb bridge chip on mega board and esp-link on rx/tx lines

Capiten, 2020-12-28:
hum..so whats the turn around? desolder the connection to the bridge?

uzi18, 2020-12-28:
This is first solution
Second is to use level shifter
And third use such connections:

Capiten, 2020-12-28:
Thank you but it didn't worked. Keeps complaining about http POST request to  http://192.168.1.96/pgmega/upload returns 404.

uzi18, 2020-12-28:
@Capiten with: RX(pin0) ----[>]---- TX pin?

Capiten, 2020-12-28:
Yes.
Managed to get it to work trough another method. Didn't moved anything, the wires are as you sugested, in fact itm is as I was doing before, but now I have a diode (1N4007) with cathode facing the esp and anode on the mega.
The way its working now is with virtual com port created with 'HW Virtual serial port' and i'm using avrdude, but since that had a reset problem, i'm using the bash script from CharlyBrok, as follows:
#!/bin/bash

CMD="avrdude -p m2560 -c stk500v2 -P net:$1:2323 -U flash:w:$2 -D -v"
CMD23="avrdude -p m2560 -c stk500v2 -P net:$1:23 -U flash:w:$2 -D -v"

timeout 1 $CMD

$CMD23
echo "Done"
exit

Usage : ./flash_mega.sh esp-e62067 /tmp/arduino_build_495949/TFT_Test_Mega_3.ino.hex

Thank you for your sugestion, but I'm able to see the serial port on esplink just fine, in fact thats how I knew I was communicating and reseting, I had prints on setup and a counting being printed out on loop.

Still don't know why all these problems with the http response =\

Capiten, 2020-12-28:
Moreover, removed the diode and still uploads without a problem.

Capiten, 2020-12-28:
I edit my comment because the virtual serial had nothing to do with it, the command to upload needs the ip, or hostname, not the com port...so many attempts I get lost ^^

uzi18, 2020-12-28:
use of diode depends on mega board design
in fact it is possible to use avrdude directly with explicity stk500v2 protocol, but this uses direct connection and not http api <- it also should work if you use my build of esp-link with mega support

Capiten, 2020-12-28:
You lost me at '..use my build of esp-link with mega support', can you please explain?

uzi18, 2020-12-28:
https://github.com/jeelabs/esp-link/issues/480#issuecomment-577457913

uzi18, 2020-12-31:
@Capiten any update?