jeelabs/esp-link #510: NodeMCU doesn't create AP after flash

Technic1975, 2020-12-13:
Hi all,

I am having real trouble flashing an AZDelivery NodeMCU WiFi Lolin V3 ESP8266 ESP-12F with esp-link, I have been trying for days with various settings but I am having no luck, below are the settings I am using as I think they should be.

[image]

After flashing and restarting the AP only shows now and again on my tablet and on inspection of the serial data the nodeMCU just keeps resetting as shown below

[image]

When I first flashed this module it appeared to work and I was able to log in at 192.168.4.1 and connect to my router but it has never worked since, I have erased and reflashed with different setting 100's of times but I just cannot get it to work.

Please could someone shed some light as to what the problem is, I think it is linked to this line  129> Flash config restore *FAILED* but I am pretty sure that    171> Flash map 4MB:1024/1024, manuf 0x68 chip 0x4016 are the correct settings for this board.

Kind regards

Jason

Technic1975, 2020-12-30:
Well I have a further development, although the AP never shows, I have discovered that if I erase the module, then flash the hello server sketch with Arduino 1.8.14 with my SSID and password details included, then flash ESP link over the top, minus the Blank.bin file, I am able to connect to ESP link on my pc over my local network with IP 192.168.0.65 and configure as I like.

I updated the firmware to NMEA link and am able to send GPS data as I had planned, the unit still won't set up an AP though if it cant connect to the specified network.

This is really frustrating as I feel I am very close, does anyone have any Ideas?

uzi18, 2021-01-02:
Related to #490

Technic1975, 2021-01-10:
Well I purchased a second module, same results! after a lot of frustration I have found that if I flash these modules very slowly at 9600 Baud rate that they work just fine with all versions of ESP link, very unstable flash chips!

closed 2021-01-10 by Technic1975