jeelabs/esp-link #515: How can disable auto reset microcontroller after tern on esp 01 module??

hamedmalekii, 2021-01-30:
Hi 
I after connect3.3  voltage to my esp 01for turn on module ,my ardunio uno board has reseted
How can this feature be disabled?

uzi18, 2021-02-01:
You can add capacitor between RESET and GND

hamedmalekii, 2021-02-03:
> You can add capacitor between RESET and GND

What  should I use the capacitor with number?
Does adding a capacitor prevent the uno from reset when the sketch is not uploaded?
Because my problem is just resetting the uno when the module ESP-01 is turned on

uzi18, 2021-02-04:
@hamedmalekii https://www.youtube.com/watch?v=AItW1tfbLTc

uzi18, 2021-02-04:
@hamedmalekii please describe your configuration and connections esp-uno

hamedmalekii, 2021-02-04:
> @hamedmalekii please describe your configuration and connections esp-uno

ok i shared  
[image]

uzi18, 2021-02-04:
@hamedmalekii OK try to add 1000uF capacitor on 3v3 and GND line

hamedmalekii, 2021-02-06:
> @hamedmalekii OK try to add 1000uF capacitor on 3v3 and GND line

not working for me 
I used 10uf capacitor between reset uno line and GPLO0 and my problem solved 
thanks for everyone

closed 2021-02-06 by hamedmalekii