jeelabs/esp-link #516: How to change UART Baud Rate

pilumalu, 2021-02-07:
Hello,

first, thanks a lot for the awesome esp-link !!
Please, how can I change the UART baud rate. Is there a hidden menue I didn`t find ?
Would be very helpful to be able to change all serial settings similar to #511 request.
Thanks !

ccutrer, 2021-02-07:
you can go to http://esp-link/console.html and use the dropdown, or manually set it to an arbitrary rate with http://esp-link/console/baud?rate=X (and http://esp-link/console/fmt?fmt=8N1 etc. for the other serial settings). You'll need to be on a 3.0 release for the latter to work.

uzi18, 2021-02-07:
Yes you can change serial settings on console subpage

uzi18, 2021-02-07:
If you need 3.0 link is somewhere in issue #513

pilumalu, 2021-02-08:
Thanks for pointing me to 3.2.47.
Just flashed and baud setting works great now !
Good work !!

closed 2021-02-08 by pilumalu