jeelabs/esp-link #518: ESP8266+Arduino BIOS flasher.

alifilhan0, 2021-02-16:
Hello, I want to use ESP8266 with a Arduino to dump and write external EEPROM chips, mostly as a project to flash EEPROMs Over The Air. Just like this......

https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&cad=rja&uact=8&ved=2ahUKEwi-gI6H2u3uAhVOVH0KHShTBHUQFjAAegQIAxAC&url=https%3A%2F%2Fwww.multirotorguide.com%2Fguide%2Frestoring-bricked-eachine-prodvr-after-failed-firmware-update%2F&usg=AOvVaw0Fwh5usXgMN4B3JnWl0hI0

I know this is not an issue actually but I don't know where to ask questions about this. The thing I want is replace the CH340G USB to UART and use the ESP8266 with Arduino pro mini to flash EEPROMs over Wifi. Is this possible? I know that Serprog supports TCP too, but how can I use it to flash Wirelessly?

uzi18, 2021-06-09:
@alifilhan0 good question, need to understand your idea correctly first

uzi18, 2021-06-09:
links to check:
https://www.multirotorguide.com/guide/restoring-bricked-eachine-prodvr-after-failed-firmware-update/
https://github.com/urjaman/frser-duino
https://github.com/flashrom/flashrom/blob/master/Documentation/serprog-protocol.txt

uzi18, 2021-06-09:
original command:
`mingw32-w64-flashrom-r1781.exe --programmer serprog:dev=COM10:115200 -c MX25L1605 --read test.hex`
this should work:
`mingw32-w64-flashrom-r1781.exe --programmer serprog:ip=esp_link_ip:23 -c MX25L1605 --read test.hex`
just change esp_link_ip for its ip adress