jeelabs/esp-link #519: gpio usage

lawashnik, 2021-02-23:
Hi. How can I programm gpio pins? Besides using serial bridge want to manage gpio's state over wifi. Is it possible? Like curl requests or something..?

uzi18, 2021-02-23:
@lawashnik Actually it is not supported

lawashnik, 2021-02-24:
I find something like solution. I can change reset pin to any and its state become high. But previous pin state doesn't become low. It becomes low only after resetting esp. Better than nothing I guess.

uzi18, 2021-02-24:
How many gpios you need?

lawashnik, 2021-02-24:
> How many gpios you need?

Only one.

uzi18, 2021-02-24:
Can add api to set/reset this pin if you need

janb14, 2021-03-12:
I have also interest in switching gpio through the webapi. I try to build an automated esp32-cam testing stand. If a firmware over OTA fails i would like to use this project to flas a default image. The problem is i need to toggle power and connect GPIO0 to GND to programm the esp32-CAM on demand.

ktostam, 2023-08-13:
The ability to control even one GPIO via API would be great.

uzi18, 2023-08-13:
It is possible. Someone should write it ;)