jeelabs/esp-link #521: Possible to choose/compile other ssid than ESP_xxxxxx ?

zuidwijk, 2021-03-28:
I'm wondering, is it possible to choose another name as ssid? Of course by comping your own firmware...
I'm search through the code yet I can't find if possible, and where (if indeed it's possible).

zuidwijk, 2021-06-08:
Yeah ... but how/where? I've search through the whole code (did also searched with grep through the code)

uzi18, 2021-06-08:
@zuidwijk here: 
https://github.com/jeelabs/esp-link/blob/cf329e8b84a55a05afa04df971d9abd9793c9130/Makefile#L24-L41

zuidwijk, 2021-06-08:
Nice! Gonna try that later today ☺️

zuidwijk, 2021-06-11:
Just tried it, unfortunately it doesn't work ....
```
# The SOFTAP configuration can be hard-coded here, the minimum parameters to set are AP_SSID && AP_PASS
# The AP SSID has to be at least 8 characters long, same for AP PASSWORD
# The AP AUTH MODE can be set to:
#  0 = AUTH_OPEN, 
#  1 = AUTH_WEP, 
#  2 = AUTH_WPA_PSK, 
#  3 = AUTH_WPA2_PSK, 
#  4 = AUTH_WPA_WPA2_PSK
# SSID hidden default 0, ( 0 | 1 ) 
# Max connections default 4, ( 1 ~ 4 )
# Beacon interval default 100, ( 100 ~ 60000ms )
#
AP_SSID ?=testname
AP_PASS ?=testname
AP_AUTH_MODE ?=0
AP_SSID_HIDDEN ?=0
AP_MAX_CONN ?=4
AP_BEACON_INTERVAL ?=100
```

AP_SSID & AP_PASS are set as they're minimum required

uzi18, 2021-06-12:
What does it mean, ap is not shown with your name or password does not work?

zuidwijk, 2021-06-12:
> What does it mean, ap is not shown with your name or password does not work?

It doesn't matter what I use, ... it keeps popping up with ESP_xxxxxx

uzi18, 2021-06-13:
Did you check after complete wipe flash?

zuidwijk, 2021-06-13:
Yeah I always wipe before flash. 
I git cloned it to another directory and then could successfully change the ssid. Yet the other settings in config.h (or .c, one off those two) now don't apply... guess have to play further with the settings.

uzi18, 2021-06-13:
make clean command is your friend, try it in old dir

So it work for you now?

zuidwijk, 2021-06-13:
Yeah I'll try that later (probably tomorrow)

I'm compiling via docker, so how do I make a "make clean" there?

mamert, 2021-06-30:
Well, the initial SSID is NOT "ESP_whatever", the initial SSID is "what was last used before you flashed esp-link".
Flashing esp-link doesn't change existing wifi settings, sadly.

So, as part of the flashing procedure, you could:
1. build a small firmware (espressif or arduino) with command to start an AP **with the creds you choose**
2. flash that
3. press reset to run that code
4. flash esp-link