jeelabs/esp-link #524: flash settings for D1 Mini Pro?

smitec11, 2021-05-07:
I'd tried to flash ESP-link to a D1 mini pro. which has 16M bytes(128M bit),
can someone tell me which settings to use?

I got it working with : 

python "../esptool/esptool.py" --port COM15 --baud 230400 write_flash --flash_freq 80m --flash_mode dio --flash_size 4MB 0x0000 boot_v1.6.bin 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin

but (off course) the module shows up as an 4MB.

which settings must I use for the 16M D1 mini pro?

uzi18, 2021-06-08:
Please provide flash chip id etc. From webserver page

smitec11, 2021-06-08:
Flash chip ID =  0xEF 0x4018

uzi18, 2021-06-08:
You can use same as for 32Mbit - 4MBytes
It is normal because espressive api limitations. 
Rest of flash is not used (from 0x400000)

closed 2021-06-09 by smitec11

veonua, 2022-09-07:
what is the final string form Wemos D1 Mini?
where can I get the tools, the release doesn't have "../esptool/esptool.py"

uzi18, 2022-09-07:
You need to install it first on your system

veonua, 2022-09-08:
[image]

at the same moment, I can flash using Arduino IDE and can't connect with 

`python2 "./esptool/esptool.py" --port /dev/ttyUSB0 --baud 230400 write_flash --flash_freq 80m --flash_mode dio --flash_size 4m 0x0000 boot_v1.6.bin 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin
`

I've tried to close Arduino IDE, another USB port, and reboot. no luck