jeelabs/esp-link #528: ESP-Link fails to flash an Arduino

ggodart, 2021-07-30:
I have ESP-Link installed on both ESP-01 and WEMOS D1, in both cases the console works fine and I can reboot a connected Arduino, however I cannot upload a new flash image to the Arduino using any of the following methods.

**AVRFLASH**
./avrflash -v 192.168.100.xx $THISFILE
Gives many errors terminating with
Error checking sync: sync abandoned after 8 attempts

**avrdude on a Raspberry PI**
sudo avrdude -v -c arduino -p atmega328p -P /dev/ttyUSB0 -b 115200 -F -U flash:w:$THISFILE:i
works fine with a local USB cable, however
sudo avrdude -v -c arduino -p atmega328p -P net:192.168.100.xx:23 -b 115200 -F -U flash:w:$THISFILE:i
fails as follows;

avrdude: Version 6.3-20171130
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/etc/avrdude.conf"
         User configuration file is "/root/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : net:192.168.100.41:23
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
ioctl("TIOCMGET"): Inappropriate ioctl for device
ioctl("TIOCMGET"): Inappropriate ioctl for device
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00

avrdude done.  Thank you.

**megaflash**
/home/pi/esplink/megaflash.sh -v 192.168.100.xx $THISFILE
Loops forever with
{ [9 bytes data]
* Closing connection 0
  Waiting for sync...
* Expire in 0 ms for 6 (transfer 0x1e2f8b0)
*   Trying 192.168.100.41...
* TCP_NODELAY set
* Expire in 200 ms for 4 (transfer 0x1e2f8b0)
* Connected to 192.168.100.41 (192.168.100.41) port 80 (#0)
> GET /pgmmega/sync HTTP/1.1
> Host: 192.168.100.41
> User-Agent: curl/7.64.0
> Accept: */*
> 
* HTTP 1.0, assume close after body
< HTTP/1.0 200 OK
< Server: esp-link
< Connection: close
< Cache-Control: no-cache, no-store, must-revalidate
< Pragma: no-cache
< Expires: 0
<

**pgm sync**
http://192.168.100.xx/pgm/sync and http://192.168.100.xx/pgmmega/sync 
both give 
sync abandoned after 8 attempts

**What I've also tried**
None of the following work;
- Reflashed bootloader on Arduino Uno
- flash a Hex file with and without bootloader
- ESP-link v3.0.14 and v3.2.47
- Arduino Uno and Mega

It seems something is broken with the sync command - any ideas what I've done wrong?
Thanks
Giles

uzi18, 2021-07-30:
what bootloader did you use for uno?

ggodart, 2021-07-30:
I just took the default from the Arduino IDE, I’m not sure how to chose a different one

G

uzi18, 2021-07-31:
@ggodart what is your configuration on "console" subpage?
```
./avrflash -v 192.168.100.xx $THISFILE
Gives many errors terminating with
Error checking sync: sync abandoned after 8 attempts
```

please attach here all output

**but first** 

- are you sure you have communication on console subpage?
You can upload one of scatches where serial is used and try to communicate.

- do you use levelshifter?

- did you tried to switch RX and TX pins?

ggodart, 2021-07-31:
Yes, the console works perfectly and I can also reset the microcontroller from esp-link so I think hardware is ok.
I’m not using a level shifter, do you think I need one?
I tried swapping tx and rx and console stopped working ( not surprisingly)

Giles

uzi18, 2021-07-31:
ok so waiting for log from avrflash -v ....

ggodart, 2021-08-02:
avrdude log attached

[avrdude_log_02-08-2021.txt](https://github.com/jeelabs/esp-link/files/6916554/avrdude_log_02-08-2021.txt)

uzi18, 2021-08-03:
asked for avrflash log

ggodart, 2021-08-03:
OOPS - sorry, avrflash.log attached, looks like there is a problem with the sync command.
I can see the lights on the mega flickering when it is attempting the connection.

[avrflash.log](https://github.com/jeelabs/esp-link/files/6922874/avrflash.log)

ggodart, 2021-08-03:
I think its connected with the pgm/sync command, if you issue it with chrome, you get "sync abandoned after 8 attempts"
here is the esp-debug log

[esp-link pgm sync log.txt](https://github.com/jeelabs/esp-link/files/6926905/esp-link.pgm.sync.log.txt)

ggodart, 2021-08-04:
I now have this working by re-flashing the bootloader on the Mega and using the megaflash script to upload my sketch onto the target. 

It appears that my older Megas and in particular clones have a bootloader that is incompatible with ESP-Link.

For the edification of any others attempting this, I followed the instructions at https://support.arduino.cc/hc/en-us/articles/360012048060-How-to-burn-the-bootloader-between-two-Arduino-Mega to reflash mega to mega
which is unclear about how to change the ArduinoISP sketch for the wires they recommend. It should read;
`
#define USE_OLD_STYLE_WIRING

  #ifdef USE_OLD_STYLE_WIRING
    #define RESET     53 
    #define PIN_MOSI	51
    #define PIN_MISO	50
    #define PIN_SCK		52

  #endif
`
Thanks
Giles

closed 2021-08-04 by ggodart

uzi18, 2021-08-04:
But you didn't said about using mega board, first posts points to uno (atmega328)

ggodart, 2021-08-05:
UNO gave same problem and same solution