jeelabs/esp-link #529: Problem for using esp link in ESP-01S

hamedmalekii, 2021-10-02:
Hello guys 
I have problem via use esp link over ESP-01S 
after write frimware I cannot find wifi in my wifi device
Any one can help me for fix this problem?
I used 1mb flash chip size and this address 
`    0x00000 -cf boot_v1.7.bin
     0x01000 -cf user1.bin
     0xFC000 -cf esp_init_data_default.bin 
     0xFE000 -cf blank.bin  `

ananyevgv, 2021-11-12:
Hello hamedmalekii, I did it with these addresses
0x00000 boot
0x1000 user1.bin 
0x7C000 esp_init_data_default.bin
0x7E000 blank.bin

mantisrobot, 2021-12-15:
I Had the same issue!

I have a ESP-01S device that reports:
flash devID: 4014h
QUAD:8Mbit
Crustal: 26Mhz

With V2.2.3, I used these settingsfor a 1Mbyte device:
0x00000 boot_v1.5.bin
0x01000 user1.bin
0xfe000 blank.bin

With V3.0.14 I have to use setting as if the device is a 512Kb device:
0x00000 boot.bin
0x01000 user1.bin
0x7e000 blank.bin
0x7c000 esp_init_data_default.bin

if I use the settings as per V2.2.3 I get the fast blink boot error!

uzi18, 2021-12-15:
Please provide bootlog from serial port (from 2.2 and 3.0)

Also please try latest beta firmware 3.1.x these old ones are not supported right now
Will try to check if something is wrong.

paulvk2, 2023-01-08:
I have the same issue with an extra problem.
I can not program the esp-01s with the flash download tool it always get the fast led flash boot load.
I can program it with NODEMCU programmer and it boots but the AP is not running/broadcasting I can load a clock program into it with the arduino ide and it works connecting to my lan and getting the time from time servers so I know the esp-01s works.
below is the serial output to my terminal
                                                                              
   239> 
                                                                      
   240> ** esp-link v3.2.47-g9c6530d
                                          
   240> Flash config restore *FAILED*
                                         
   240> CONN led=0
                                                            
   240> SER led=14
                                                            
   242> Wifi init, mode=AP
                                                    
   244> Wifi uses DHCP, hostname=esp-link
                                     
   248> "ip": "0.0.0.0"
                                                       
   250> "netmask": "0.0.0.0"
                                                  
   253> "gateway": "0.0.0.0"
                                                  
   255> "hostname": "<null>"
                                                  
   258> sleep enable,type: 2
                                                  
   261> Httpd init, conn=0x3fff4128
                                           
   264> No user file system found!
                                            
   267> Serbridge pins: reset=12 isp=13 tx_enable=-1 swap=0
                   
   281> Reset cause: 6=external
                                               
   281> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
           
   282> Flash map 1MB:512/512, manuf 0xD8 chip 0x4014
                         
   286> ** esp-link v3.2.47-g9c6530d: ready, heap=18960
                       
   291> SNTP timesource set to us.pool.ntp.org with offset 0
                  
   296> initializing user application
                                         
   300> Waiting for work to do...
                                             
   303> mode : softAP(ea:db:84:da:02:e6)
                                      
   306> add if1
                                                               
   308> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
  
   316> bcn 100
                                                               
 15261> Wifi check: mode=AP status=255

paulvk2, 2023-01-08:
Ok I have esp8266EX 4MB and it works on it but not on the ESP-01S.

paulvk2, 2023-01-08:
I went through the earlier versions and it stops working after 2.1.7