jeelabs/esp-link #532: System is periodically reset by WDOG when baudrate is set to 2400

cz8pru, 2021-10-07:
I have usecase where I need to communicate via UART at 2400. It is working till my device is reset. After that, it will finish in infinitive loop. it start and it is reset by WDOG due to the fact, that messages to UART TX from ESP-LINK are written too slowly.

zalexua, 2022-09-01:
I have the same issue. The reboot loop is 3.7 seconds long.

Have you found any solution?

cz8pru, 2022-09-02:
I did not find solution neither work around. 

This is bug which needs to be fixed.

"---------- Původní zpráva ----------

baudrate is set to 2400 (#532)

(https://github.com/jeelabs/esp-link/issues/532#issuecomment-1234548512), or
unsubscribe

"

zalexua, 2022-09-02:
Yesterday later I found that there is a menu "Debug log" where you can turn off logging to uart0 and it resolved the issue.
See
[image]

cz8pru, 2022-09-02:
Good job, nice workaround!
---------- Původní e-mail ----------

Komu: jeelabs/esp-link ***@***.***>
Kopie: cz8pru ***@***.***>, Author ***@***.***>
Datum: 2. 9. 2022 8:42:10
baudrate is set to 2400 (#532)
"

debug-log-off 
(https://user-images.githubusercontent.com/273254/188074764-1f819d23-b36b-415c-99fe-71f27a555979.png)