jeelabs/esp-link #535: Homeautomation connection for mqtt, wemo and hue?

hjahend, 2021-12-07:
Hi all, I use Tasmota based and other devices for my home automation for many years already.
I already also use the reduced version of esplink for transparent serial communication via wifi with other devices (homautomation software on a pc  and http commands via webbrowser)

Wondering if this extended esplink version could send and receive (relay function) MQTT, WEMO /HUE messages via a serial port on my home automation controller?
Questions:
Is it at all possible?
What message format as inputs to this software? Same for outputs (messages back)?
Do I have to alter this esplink software (if so what and how)?
How to bind it all together and check it??

uzi18, 2021-12-07:
What is extended version of esp-link?
Do you want to use esp-link as a bridge mqtt to your PC via serial port?

hjahend, 2021-12-08:
Yes. refer to these webpages & download link (simple version)
https://github.com/jeelabs/esp-link

https://atadiat.com/en/e-esp8266-esp32-wifi-serial-bridge-converter/
https://atadiat.com/en/e-esp8266-esp32-wifi-serial-bridge-converter/#

Understand that with the extended version much more is possible?

I already use this MQTT broker on a Nodemcu now for many years:
https://github.com/martin-ger/uMQTTBroker

The problem is that this MQTTbroker gets flooded with messages and breaks down. So I need something else/better, preferably on an ESP Nodemcu.

Greetings