jeelabs/esp-link #537: DT-06 Problem

leeyome, 2021-12-17:
Esp-link-v3.0.14 was applied to DT-06.
I wrote Firmware according to the installation manual, but the chip became a brick. (Now I have 2 DT-06 bricks)

DT-06 is a product using an ESP-M2 (8285) chip, 
and a CP2102 chip was used for flash writing.

The writing speed was 115200 and 460800, and so on.

I applied it to another ESP8266 chip just in case, and it was properly written.

What's the problem?

leeyome, 2021-12-17:
When I connect to the PC with CP2102, only the following messages continue to appear.

---------------------------------------------------------------------------------------------------
load 0x40100000, len 29960, room 16 
tail 8
chksum 0xfd
load 0x3ffe8000, len 2140, room 0 
tail 12
chksum 0xf9
ho 0 tail 12 room 4
load 0x3ffe8860, len 9328, room 12 
tail 4
chksum 0x0a
csum 0x0a
r[1B]?1B]륿[00]꾛?b[0C]$똁r|[04]lB乶?03]#|?04][0C]??03][12]c液o?03][12]no?10][02][00][03]꼕o?00][04][0C]럩[1C]l뙊l$똾l꾩꼕 [02[03]N?00]꾛?b[0C]d[0C]l뢳p[04][0C][02][02]뙚헱B[04]l[04]똠?0E]drl꾟[03]?13]긏??먵n|?0C]?럠d`[03]릖[03][12]n?13]?03]r{o[13]o쐍?0C]???1C][1B]?03][04]b[0C]b[1B]on[1F]?0C][0C]B|?12]{rN[1F]?똡`[03]#굛o찰[12]嗇뢳n[12]n쏯?04]???1C][1C]?[0C]"#[12]no[1E][0C]"?[1B]rso>?0C][04]뢵$?1B]N仇[12]嗇뢳n[12]n쏯?04]?쒅쐩썑[0C][0C]"[0C]b[1B]on[1F]?04]c|?1B]rso[1E]뭠꼕 [
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 29960, room 16 
tail 8
chksum 0xfd
load 0x3ffe8000, len 2140, room 0 
tail 12
chksum 0xf9
ho 0 tail 12 room 4
load 0x3ffe8860, len 9328, room 12 
tail 4
chksum 0x0a
csum 0x0a
~s[12]?1B]륿[00]뙚?b[0C]l꼉s|[0C]l"乶?03]c|?0C][04]??[1B]#任N?03][12]no?10][03][00][03]똪o?0C][04]룎[04]l꼷[00]l$?1F]꼕뚥똡`[03][03]o?00]꽳[12]cB[04]$[04]l럾[0C]B[04][03][03]뙙?b[0C]l[0C]??0E]l{$꾟[03]?1B]괤?{궅N|?0C]?1C]?$`[03]?[02][12]N?1B]?03]rrN[1B]n??04]???1C][12]?03][0C]c[04]b[13]no>?04][04]b|?1B]rrN[1F]뭠똡 [02]b긿o沁[13]?02]{ro[12]n??04]???1C][1C]?0C][0C]#b[12]no>[04]"??rrn[1F]?0C][0C]?0E]l?1B]o綎[13]?02]{rN[13]n??04]?複쐪{?0C][0C]c[0C]b[12]nn[1F]?04]B|?13]{{o>??1C]
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

uzi18, 2021-12-17:
Please try to adjust proper baud rate, it looks like not 115200, than past log here

leeyome, 2021-12-19:
> Please try to adjust proper baud rate, it looks like not 115200, than past log here

baud rate is 115200 thnx

uzi18, 2021-12-20:
this is 115200: 
```
ets Jan 8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 29960, room 16
tail 8
chksum 0xfd
load 0x3ffe8000, len 2140, room 0
tail 12
chksum 0xf9
ho 0 tail 12 room 4
load 0x3ffe8860, len 9328, room 12
tail 4
chksum 0x0a
csum 0x0a
```

but rest is different, check different baud rate