jeelabs/esp-link #540: ESP8285(ESP-M3/ESP8285H16) 16Mbit chip flash issue

rootmax84, 2022-02-04:
So, i got another ESP-M3, it's definitely 2mbyte in size. esptool said its QUAD:16Mbit.

Trying to flash 3.2.47 with addresses of 16mbit:
0x00000 boot_v1.7.bin
0x01000 user1.bin
0x1FC000 esp_init_data_default.bin
0x1FE000 blank.bin
Fail. Just Blinking. No debug in uart

Trying to flash 3.2.47 with addresses of 8mbit:
0x00000 boot_v1.7.bin
0x01000 user1.bin
0xFC000 esp_init_data_default.bin
0xFE000 blank.bin
Fail. Just Blinking. In uart we got rf_cal 0 =0x05 is 0xff

Trying to flash 3.2.47 with addresses of 4mbit:
0x00000 boot_v1.7.bin
0x01000 user1.bin
0x7C000 esp_init_data_default.bin
0x7E000 blank.bin
Success. But without space for web pages it's sad

SPI mode DOUT, flash download tool 3.9.2

Have no idea why. Seems need to update esp-link binaries to support new esp chips :)

uzi18, 2022-02-04:
please provide boot log from ESP

rootmax84, 2022-02-04:
> please provide boot log from ESP, try to use my full flash image

What kind of full image? Where i can get it?

Boot on 8mbit and 16 says it's 512kb, but esptool says 16mbit, am confused :)

[image]

uzi18, 2022-02-04:
It should be 16Mbit (2MB) flash

uzi18, 2022-02-04:
https://github.com/jeelabs/esp-link/blob/cf329e8b84a55a05afa04df971d9abd9793c9130/Makefile#L158-L168

uzi18, 2022-02-06:
@rootmax84 in short even if it is 2MB flash chip, only 1MB may be used as program (thats why it is 512+512 for OTA support), rest could be data only

uzi18, 2022-02-07:
@rootmax84 please try to flash at 0x80000 first blank.bin and next esp_init_data_default.bin file and try show me boot log like before.

rootmax84, 2022-02-10:
@uzi18 
Like this?
0x00000 boot_v1.7.bin
0x01000 user1.bin
0x80000 blank.bin
0x1FC000 esp_init_data_default.bin

uzi18, 2022-02-16:
@rootmax84 almost 
`0x80000 blank.bin`
and next 
`0x80000 esp_init_data_default.bin`

rootmax84, 2022-04-09:
@uzi18 

2nd boot version : 1.7(5d6f877)
SPI Speed : 80MHz
SPI Mode : DOUT
SPI Flash Size & Map: 4Mbit(256KB+256KB)
jump to run user1 @ 1000

rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:1, boot mode:(3,5)

load 0x40100000, len 2592, room 16 
tail 0
chksum 0xf3
load 0x3ffe8000, len 764, room 8 
tail 4
chksum 0x92
load 0x3ffe82fc, len 676, room 4 
tail 0
chksum 0x22
csum 0x22

this logs infinitely.

info rom esptool

esptool.py v3.2
Found 2 serial ports
Serial port COM6
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...
Detecting chip type... ESP8266
Chip is ESP8285H16
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: 70:03:9f:42:3c:f6
Uploading stub...
Running stub...
Stub running...
Manufacturer: a1
Device: 4015
Detected flash size: 2MB
Hard resetting via RTS pin...

rootmax84, 2022-04-11:
@uzi18 

This is definitely an ESP8285H16 chip, I opened the protective shield from the chip, the markings match, it's 2mb. Why after the firmware in the console he writes SPI Flash Size & Map: 4Mbit (256KB + 256KB) is not clear to me. It works accordingly only if it is flashed as a 512kb chip. I would be grateful for help

closed 2022-04-11 by rootmax84

uzi18, 2022-04-11:
What was wrong version?