jeelabs/esp-link #543: ESP8266-01s with CH340G USB

msmthng, 2022-02-26:
I was able to flash using a modified CH340.

[image]

After reconnecting the device (with an unmodifed) USB Adapter, the blue led on the ESP8266 is flashing. No WiFi is available.

Does anybody have an idea what could be the reason for that.

thx for support.

msmthng, 2022-02-27:
same behavior with both tries.

uzi18, 2022-02-27:
@msmthng do you have serial log ?

msmthng, 2022-02-27:
how i can collect the serial log?

uzi18, 2022-02-27:
Use terminal application like putty with log to file feature enabled.

uzi18, 2022-02-27:
You can write to me on gitter.im

msmthng, 2022-03-01:
logoutput:

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.03.01 11:34:41 =~=~=~=~=~=~=~=~=~=~=~=
„ãû“pÃ'g€llllÇcì‡ãâ›`sd$œž|Œlà|Œlìc|ŽÃä“;“cŒcŒûg'Ÿlngœãäbxäl;l{lpûgà‚lc'ã|ÄlŽ<„cŒûogçl„‡l`ng$`'{‡“›gã{l xónà{Œœœãàcoã|ì‡ŽcŒûo'çl o'‡lo;‡›ÛgŒl`xûoà;ŒœœãàŒŒœ8xgã|„‡Çc„ûgnî‡l`no$`'{’›ob„l r›ÛgcŒl`{l„Œoœ„ûgoŸ„ŸãÄ{'ogÄdŒccsãŒœìŽ‡l;l‡“dä“ÛƒŒÇ$d${$›lllgœãŒl›ldl‡’d„˜'à‡nì›o~ŒŒ$dà€bÇ‡'Ÿc“ŒŽcsb›„#cll“›oßnŒ“ãgdŸ|c$d`;lÄãòÛxƒo'€dl$lcìâãÛ`{$lœŸ|Ädà|„lä#|‡ƒìÛs›b„c„ò'oßdg'œãäcpäÇdsdrdpò'àƒlcoã<Œd‡|Äc„ûgoîdŒÇ$`˜g'l`orÇ›’'ãrd`8ûgà;„œÜãàb'ã|ä‡Žc„û'oîd`˜goŽdgsÛ“oŒ$`pûgàsÄœœãàŒÄœppnâ|ŒÇ‡#ŒóngïÇ$`o'l`o;‡›ÛgcŒl`{Û“o#Œ$`sdÄÄgœŒóogžŒßãŒrognŒlŒb‡#;ãÄœä‡ÇdsdŽ›lìÛ“ƒ„l$l~;l‡’ddd'œã„d’dldŽ›lŒoàÇgä’g„Œllà€cÇoßb›„#{c’ÄÇc#l$›Û'Ÿo„›ãnlž|cl$`sd„ãû“pÃgg€llllÇcì‡ãâ›`sd$œž|Œlà|Œlìc|ŽÃä“;“cŒcŒûg'Ÿlngœãäbxäl;l{lpûgà‚lc'ã|ÄlŽ<„cŒûogçl„‡l`ng$`'{‡“›gã{l xónà{Œœœãàcoã|ì‡ŽcŒûo'çl o'‡lo;‡›ÛgŒl`xûoà;ŒœœãàŒŒœ8xgã|„‡Çc„ûgnî‡l`no$`'{’›ob„l r›ÛgcŒl`{l„Œoœ„ûgoŸ„ŸãÄ{'ogÄdŒccsãŒœìŽ‡l;l‡“dä“ÛƒŒÇ$d${$›lllgœãŒl›ldl‡’d„˜'à‡nì›o~ŒŒ$dà€bÇ‡'Ÿc“ŒŽcsb›„#cll“›oßnŒ“ãgdŸ|c$d`;lŒãó’8ƒoo€$dld‡#äãã“`;ldœŸ|Œ$à|Ädìc<‡ƒì’{ÛcÄ#ŒóogžlgoœãìcpìŽ${$sdxóoàƒd#gã|„$Ç|Œ#Äónoï$Œl ˜god`gsÛ“oãs$`pûgàsÄœœãàcnâ|ìÇ‡#Œóngï$`no$'{“›o„l pónà{ŒÜœâà„ŒÜxpoã|Œ‡ÇcŒûo'çl ˜gnd`osŽ›“ncŒd`{“›oc„l s$ŒŒ'œŒûo'ŸŒŸã„sgno„l„cÇcrãŒÜä‡${lÛlä’›ÃÄ‡lldsdÇ“$ldoœãŒ$“$l$ÛlŒgàgä“'Ä„dlàÀc‡gžcÛÄ‡c;c“ŒŽcbdlÛ“nŸgÄ’ãolŸ<cdl {$„ãû“pÃ'g€lldlÇcìÇãâ›`sd$œž|Œlà|Œlìc|Ãä›;“cŒcŒûg'Ÿlngœãäbxä‡l;l{lpûgà‚lcnã|Äl<„cŒûo'çl„‡d`ngl`'{‡“›gã{l xónà{Œœœãàcgã|ì‡ŽcŒûg'çl o'‡lo;Ç›Û'Œd`8ûoà;„œœãàŒŒœ8xgã|„‡Çc„ûgoî‡l`no$`'{’›ob„l r›Û'cŒd`;l„„oœ„û'oŸ„ŸãŒ{nogÄdŒccsãŒœìŽ‡l;l‡“dä›ÛƒŒÇ$d${$›lllgœãŒl›ldl‡“dÄ˜'à‡nì›o~ŒŒldà€bÇ‡'Ÿc“ŒŽcsb›„bcld“›gßnŒ“ãgdŸ|#$d`;lÄãû›xƒng€dldlcìŽãã›`s$lÜŸ<„là<Œläc|‚ì›r›#Œc„ûgoŸdogœâäc8ä‡lrd;lpû'àƒlcoâ<Œd|Äc„ûgnçdÄ‡d`ogl n;‡“Ûgã;l`xóoà;Œœœãà#gã|äÇc„ûgnçl`˜gnŽlorŽ›’nŒd`pûgàr„œÜãàŒ„œppnã|Ä‡‡cÄò'oï‡d`ogl n{‡“›gcÄl`s›’ncŒd`rlÄ„oÜÄònoßÄŸãŒ;ognŒ$Œb‡#{â„œì‡lsdÇ“$ì›’ƒŒŽldl?{l‡’ddlgœâ„l’d$dÇ“$ŒØnà‡oìÛgŒŒl$à€cŽÇnŸ#›Œ#scÛ„‡ccld“ÛgžoŒ›â'$ß|bld`rd„ãû“pÃ'g€lldlÇcì‡ãâ›`sd$œž|Œlà|Œlìc|ŽÃä“;“cŒcŒûg'Ÿlngœãäbxä‡l;l{lpûgà‚lcnã|ÄlŽ<„cŒûogçl„‡l`ng$`'{‡“›gã{l xónà{Œœœãàcgã|ì‡ŽcŒûo'çl o'‡lo;Ç›ÛgŒl`xûoà;ŒœœãàŒŒœ8xgã|„‡Çc„ûgoî‡l`no$`'{’›ob„l r›Û'cŒl`{l„Œoœ„ûgoŸ„ŸãÄ{'ogÄdŒccsãŒœìŽ‡l;l‡“dä“ÛƒŒÇ$d${$›lllgœãŒl›ldl‡“d„˜'à‡nì›o~ŒŒ$dà€bÇ‡'Ÿc“ŒŽcsb›„bcll“›oßnŒ“ãgdŸ|c$d`;lŒâó’pƒgo€$dld‡#äãã“`;ldœŸ

any setting to change in addition (beside comport, baudrate 1152000)?

uzi18, 2022-03-01:
115200,230400, text should be visible there

profroyk, 2022-04-05:
I have the same problem w/ and ESP-12 and tried several of the 3.2 releases.  It looks like the flash worked, the LED blinks about 1/second but there is no ESP-xxx AP to connect to.    I tried two different NodeMCUs - same result.  I loaded the Example skets that sets up an AP on the ESP8266 and was able to connect to it so I think the WiFi circuitry and antenna is OK.

I tried getting a log at 115200 and have garbled outputs much like the earlier example.

So, a couple of questions:

o Is there a way to connect esp-link to my WiFi network without connecting to the not-visible esp-link AP?

o What can I do to get a better log?  The putty display shows very quick lines of text which I can't read and can't be logged successfully at 115200.  There are some text characters but not enough to make sense of. small sample:
d`N{▒▒▒n
▒{$`p▒n▒
s▒ܜ▒
bn▒|쎏▒c▒▒no▒
l ▒n▒{lor▒▒▒N
▒$`p▒n▒
{▒ܜ▒▒{▒bo▒|▒▒▒▒c▒|~▒o▒▒l ▒N
l or▒▒▒N
b▒$`rےo
#▒l`rd▒▒N▒▒p~▒o▒▒▒▒ron▒
$▒b▒b{ㄜ쎎#$`▒▒$▒▒▒▒
▒l$l~r$▒▒$l$o▒▒▒l▒dl$▒▒$▒▒o▒n▒▒▒܄▒$l▒"
▒{b▒b▒▒▒crc▒▒▒ob$l▒▒▒▒▒▒n$▒▒▒b
$`r$▒▒▒p▒on▒$2$l
$▒p$o|
▒▒`sl$ܟ|▒d▒|
▒
$▒c|▒▒▒▒{▒#▒
B▒▒on▒lon▒▒#bp▒▒l{lrlp▒o▒
▒l

o What else can I try?

yesyesuk, 2022-04-05:
@profroyk The bootloader messages are sent at a baud rate of 74880. Try to receive the boot log at that rate to read it.

uzi18, 2022-04-05:
did you tries to flash blank.bin at 0x80000
and next esp_init_data_default.bin 0x80000

profroyk, 2022-04-05:
I'll try it when I get home tonight but are those addresses correct?  They
are both the same address so wouldn't esp_init_data_default overwrite
blank.bin?

Roy

uzi18, 2022-04-06:
blank.bin is to erase data and esp_init_data_defaults is to init this memory with defaults.

profroyk, 2022-04-06:
Thanks.  I put the esp_init_data_defaults at 0x80000.   It didn't make any
difference -  Same problem.  Here's a snippet of the console log:

2nd boot version : 1.6
  SPI Speed      : 80MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 4Mbit(256KB+256KB)
no GPIO select!
jump to run user1 @ 1000

rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2408, room 16
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0
tail 8
chksum 0xd8
csum 0xd8

I think I may know what's happening but I don't know what to do about it.
I tried running esptool to do the flashing instead of the ExpressIf utility
and got an error when trying to write user1.bin:

C:\Users\roykr\psuproj\esp-link\esptool>python esptool --port COM7 --baud
11520 write_flash --flash_freq 20m --flash_mode qio --flash_size 32m 0x1000
user1.bin 0x0000 boot_v1.6.bin  0x3fc000 esp_init_data_default.bin 0x3fe000
blank.bin
WARNING: Flash size arguments in megabits like '32m' are deprecated.
Please use the equivalent size '4MB'.
Megabit arguments may be removed in a future release.
esptool.py v3.2
Serial port COM7
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying
again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
WARNING: Detected crystal freq 20.89MHz is quite different to normalized
freq 26MHz. Unsupported crystal in use?
Crystal is 26MHz
MAC: 58:bf:25:d9:e6:e9
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00051fff...
Flash will be erased from 0x00000000 to 0x00000fff...
Flash will be erased from 0x003fc000 to 0x003fcfff...
Flash will be erased from 0x003fe000 to 0x003fefff...
Compressed 330756 bytes to 244619...
Writing at 0x00001000... (6 %)Traceback (most recent call last):
  File "C:\Program
Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3312.0_x64__qbz5n2kfra8p0\lib\runpy.py",
line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Program
Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3312.0_x64__qbz5n2kfra8p0\lib\runpy.py",
line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\roykr\psuproj\esp-link\esptool\esptool\__main__.py", line
11, in <module>
    esptool._main()
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\esptool.py",
line 5136, in _main
    main()
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\esptool.py",
line 4602, in main
    operation_func(esp, args)
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\esptool.py",
line 3873, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\esptool.py",
line 154, in inner
    return func(*args, **kwargs)
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\esptool.py",
line 918, in flash_defl_block
    self.check_command("write compressed data to flash after seq %d" % seq,
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\esptool.py",
line 495, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\esptool.py",
line 458, in command
    self.write(pkt)
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\esptool.py",
line 421, in write
    self._port.write(buf)
  File
"C:\Users\roykr\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\serial\serialwin32.py",
line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

Apparently esptool had trouble writing to that segment.  I tried this on
several NodeMCU's and got the same failure.  I tried writing all of the
other segments and that worked.  The only write that failed was the one to
user1.bin.  I tried writing user2.bin and had the same problem.  The write
always hangs up 6% of the way in.
o Can you think of a reason why this would be happening?  Have you seen
this before?
o Can I put user1.bin somewhere else in the flash?

Thanks
Roy

profroyk, 2022-04-06:
Breakthrough.

I flashed user2.bin to 81000 and esp-link was able to boot to user2.bin.  I
can connect to the ESP_DE9E00B network (but not every time) but now I'm
running into configuration problems.
Scanning for networks doesn't return anything and when I enter my SSID and
p/w I get a message saying message saying Error Switching Networks.

Here's the log from putty after a bunch of garbage characters which I
assume is from reset.
   121> ** esp-link v3.0.14-g963ffbb
   121> Flash config restore *FAILED*
   121> CONN led=0
   122> SER led=14
   124> Wifi init, mode=AP+STA
   126> Wifi uses DHCP, hostname=esp-link
   130> "ip": "0.0.0.0"
   132> "netmask": "0.0.0.0"
   134> "gateway": "0.0.0.0"
   137> "hostname": "esp-link"
   140> sleep enable,type: 2
   143> Httpd init, conn=0x3fff3820
   146> No user file system found!
   149> Serbridge pins: reset=12 isp=13 swap=0
   162> Reset cause: 6=external
   162> exccause=0 epc1=0x0 epc2=0x0 epc3=0x0 excvaddr=0x0 depc=0x0
   162> Flash map 4MB:512/512, manuf 0x20 chip 0x4016
   167> ** esp-link v3.0.14-g963ffbb: ready, heap=21368
   172> SNTP timesource set to us.pool.ntp.org with offset 0
   177> initializing user application
   180> Waiting for work to do...
   183> mode : sta(58:bf:25:d9:e0:0b) + softAP(5a:bf:25:d9:e0:0b)
   189> add if0
   191> add if1
   192> dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
   201> bcn 100
 15142> Wifi check: mode=AP+STA status=0
 15143> Enabling/continuing uart log
 30143> Wifi check: mode=AP+STA status=0
 30143> Enabling/continuing uart log
 45143> Wifi check: mode=AP+STA status=0
 45143> Enabling/continuing uart log

So, one gigantic step forward and now one back.

Thanks for your previous and (hopefully) continuing help - much appreciated
Roy

uzi18, 2022-04-06:
try this file, but decompress it with 7zip tool 3x, it is 3 times compressed zip,gzip,tar
[esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.tgz.zip](https://github.com/jeelabs/esp-link/files/5889597/esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.tgz.zip)

try first:
```
python esptool --port COM7 --baud 115200 write_flash --flash_freq 20m --flash_mode dio --flash_size 32m 0x1000
user1.bin 0x0000 boot_v1.7.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin 
```
check if it work
2nd check this one:
```
python esptool --port COM7 --baud 115200 write_flash --flash_freq 20m --flash_mode dio --flash_size 32m 0x1000
user1.bin 0x0000 boot_v1.7.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin 0x80000 esp_init_data_default.bin 
```

profroyk, 2022-04-06:
I have success.  I can connect to my home ssid.  I have an ip address.  I
think I'm good to go forward now.

Here's what it took:
o The python esptool failed the same way.  With a timeout when trying to
write user1.bin.  I tried both of your suggestions and then some.  Nothing
worked.  Couldn't get past the timeout at 6%
o I switched to NodeMCU flasher and flashed v3.2.47.alpha.  w/ boot_v1.7 at
0x0000, user1.bin at 0x1000, esp_init_data_default.bin at 0x3fc000 and
blank.bin at 0x3fe000.  I also flashed user1.bin to 0x81000 (which I'm
guessing is the backup partition).
o The ESP_Dxxx ssid showed up and I was able to connect.  After switching
to STA+AP I was able to scan (something I couldn't do w/ the 3.0.14
version) and connect to my home network.
o I opened a browser page to the new IP address which brought up the
esp_link page and all looks good:

[image: image.png]

Just tried it on a 2nd NodeMCU, this time with user1.bin at 0x1000 and
user2.bin at 0x81000.  Was able to join the ESP AP point, connect to my
home network, and get the esp-link web site at its assigned IP address.
Boot source was user1.bin - Yay!  I think we're done.  My conclusions:
o esptool is not as patient as the NodeMCU flasher.  No timeouts, all
partitions got written to the device
o the later release did something to fix the problems I was having
o I couldn't have done this w/o your support.  Thanks

Roy

bgrigoriu, 2022-06-19:
Hi everyone,
I have exactly the same problem and the fashing of the 3.2.47 alpha did not fix the issue for me. I am using the Expressif donwload tool. 

here are the outputs for the 3.2 and 3.0 versions

2nd boot version : 1.7(5d6f877)
SPI Speed : 40MHz
SPI Mode : QIO
SPI Flash Size & Map: 4Mbit(256KB+256KB)
jump to run user1 @ 1000

rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2592, room 16 tail 0 chksum 0xf3
load 0x3ffe8000, len 764, room 8 tail 4 chksum 0x92
load 0x3ffe82fc, len 676, room 4 tail 0 chksum 0x22
csum 0x22

And  

2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 4Mbit(256KB+256KB)
no GPIO select!
jump to run user1 @ 1000

rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 2408, room 16 tail 8 chksum 0xe5
load 0x3ffe8000, len 776, room 0 tail 8 chksum 0x84
load 0x3ffe8310, len 632, room 0 tail 8 chksum 0xd8
csum 0xd8

Can anyone explain what is the issue and propose a way to fix for a newbee like me ?

uzi18, 2022-06-19:
First try with DOUT flash mode

bgrigoriu, 2022-06-19:
Hi, Thanks 

Tried, does not work better. Flassing goes well with eithr DIO or DOUT (ESP flashing tool 3.9.2) but when booting it goes in the same loop. 
Same serial output as before

bgrigoriu, 2022-06-19:
I really do not understand what is happening. 
If I read well the log:
- it launches the bootloader which seems ok 
- i do not understand well the " no GPIO select !" but nevertheless it handover to user code @ x01000
- then it load three  "things "  then reboots ?
why so ?
I may be dumb but I am at the very begening and want to learn.

uzi18, 2022-06-19:
What is your flash size?

uzi18, 2022-06-19:
If it ia 32Mbit, you can try to use full flash attached in my previous posts

bgrigoriu, 2022-06-19:
No it is a plain 8Mbit . 
It worked  a few months ago and I really do not understand the issues. 
Can you give me the link to the " full flash" ?
Thanks !

uzi18, 2022-06-19:
https://github.com/jeelabs/esp-link/issues/543#issuecomment-1090598270

bgrigoriu, 2022-06-19:
Tried to flash the 3.2.47 but stil the same issue
I also tried to flash user2.bil  to 81000 . Flash ok but then same issue. 
Could it be because of the  expressif download tool 3.9.2 ?? this is the only thing a changed from last time ?
I really do not understand why

uzi18, 2022-06-19:
Use esptool

bgrigoriu, 2022-06-19:
I wil try . Never used before.

bgrigoriu, 2022-06-19:
Unfortunately I have a strong disability ( I am a windows guy)

bgrigoriu, 2022-06-20:
Hi,

I was able to use esptool for win
Works fine 
Some advancement:
Flash ok but the ESP network does not show up consistently and i cannot connect to it al all.

CRSREE, 2022-09-26:
hi,

Could you manage to fix this problem?

profroyk, 2022-10-11:
Tried flashing in two steps. First blank.bin to 0x80000 and then
esp_init_data_default.bin  go 0x80000.,
same result.  Fast blinking green LED, no AP on either my phone or my
laptop.

Roy

uzi18, 2022-10-12:
@profroyk any debug log? try to find me on gitter.im