jeelabs/esp-link #544: Use ESP-Link to upload program to Teensy4.1?

JackBlair87, 2022-03-28:
Hello everyone!

Does anyone think it is possible to run esp-link on an ESP32 connected over serial to a Teensy4.1, that can upload a program to the teensy? I know this library works for Arduino and other boards, but I understand that Teensy uses a different flashing system and is harder to program. Currently I have a .bin file to upload the Teensy/ESP connection set up over serial.

Thanks for all your help.

uzi18, 2022-03-28:
This teensy uses USB as a programming interface and esp-link is a serial bridge.
We need to ask @PaulStoffregen if serial programming is possible and how to proceed.
The esp-link project does not support ESP32 currently.

PaulStoffregen, 2022-03-28:
Teensy does not support hardware serial programming.  Only USB (HID protocol, not USB serial) is supported.

PaulStoffregen, 2022-03-28:
Yes, that program can be run on any PC or Linux SBC.  Or the USB host on Teensy 3.6 or 4.1 can be used with this: https://github.com/PaulStoffregen/USB_Tester