jeelabs/esp-link #545: TTL > RS232 - undesirable telnet behaviour

longmover, 2022-04-01:
I'm trying to get a wireless RS232 set up working, I've hooked up a MAX3232 TTL-RS232 chip which works but i cannot get telnet to behave properly.  I either get extra line feeds or ^@ sent with each new line.  I've tried all cominations of settings in PuTTY but no joy.  Is there a way to adjust/tweak the control characters being sent via esp-link?  It works fine for TTL-TTL, am i missing something?

uzi18, 2022-04-01:
Are you sure it is sent by serial?

longmover, 2022-04-03:
I'm not sure what you mean, the end point is an RS232 serial port and that's conected to rx/tx on the MAX3232 which is connected to the swapped pins on the D1.  It does connect and interact with the RS232 port, i can use it and it's almost perfect except for the extra characters.  The key options in PuTTY are to deslect 'Return key sends Telnet New Line instead of ^M' and to set both Local echo and Local line editing to Force off.  This addresses the double line feed issue but triggers a bell on each new line and seems to send ^@.  Toggle the 'Return key sends Telnet New Line instead of ^M' stops the bell and ^@ but causes a double line feed.

Connecting direct TTL to TTL (i've tried it on a Pi) works without the extra line feed, i'm somewhat puzzled.  What i haven't tried yet is using standard/non-swapped rx/tx pins, i'll have a go and see if that's any different

longmover, 2022-04-04:
Sam behaviour with the default rx/tx pins

longmover, 2022-04-04:
Packet capture of the PuTTY telnet session shows the client sending \r or \r\n if the 'Return key sends Telnet New Line instead of ^M' is enabled