jeelabs/esp-link #546: Flash an attatched ESP32?

Capiten, 2022-04-21:
Hi
I managed to have an esp12 working as serial bridge with an esp32, even the reset was working ok, but I couldn't upload via arduino IDE. I created an virtual com port and when I choose that com port on arduino IDE, I could see the esp32 serial communications on arduino ide serial monitor just fine, but if I tried to upload, I always got the error on arduino IDE: Failed to write to target RAM (result was 01070000).
The esp32 is connected to an UMH3N to avoid having to press boot and reset in every upload, its working perfectly trough traditional ftdi cable, as long as I have the RST to RTD and DTR to DTR connected (besides ground and TX/RX connections).

On esplink pin assigments and hardware connectios I have:
ISP/flash pin to gpio14 and this was connected to DTR pin on UMH3N.
Reset on GPIO12.
As I said I can see the serial communications on both esplink and in an virtual com port when created pointing at the esplink IP.
I fear this might be an issue on the arduino IDE side, or pherhaps what the esplinks does when receiving an arduino ide upload order does not represent what the esp32 expects to actually upload a new code.

Did anyone manage to flash an attached esp32 with virtual com port pointing to an esplink IP?
I know I could use basic OTA or something like that, but I'll have to flash boards outdoors that do not have and cannot have the extra OTA code, it has to be a real wireless upload trough a "dongle", esplink shoul be perfect.

Thank you all.

akohlsmith, 2022-11-27:
I also have this question - ESP-01 has two free GPIO, one is target reset by default, the other is "wifi connected" LED; is it possible to have control over this second GPIO to act as the BOOT pin on ESP32?

Taking it one step further, perhaps adding a control command or MQTT topic which can reset the target in boot mode so I can pipe esp-prog output through it to reprogram the target easily.

ammaree, 2023-12-27:
+99 for help to flash an esp32 devboard through an esp01 running esp-link...

Please?