jeelabs/esp-link #547: Release Artifacts not working

EduardTheThird, 2022-04-22:
I am unable to download the compiled images from the release page.
Is there a way to get the Release 3.0.14 image?

Thanks

FernandoGarcia, 2022-04-25:
The browser is blocking but you can allow the download manually clicking in download icon in toolbar.

uzi18, 2022-04-25:
First of all use my builds of 3.2.x version it has lots of fixes. 
Try to find on latest updated issues.

FernandoGarcia, 2022-04-25:
Are you talking about this? https://github.com/jeelabs/esp-link/files/5889597/esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.tgz.zip

uzi18, 2022-04-26:
You can try this but it is triple compressed

EduardTheThird, 2022-04-26:
> Are you talking about this? https://github.com/jeelabs/esp-link/files/5889597/esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.tgz.zip

This worked for me, thank you.

EduardTheThird, 2022-04-26:
Tried FireFox, the download work from the release. Chrome blocks it for some reason, just tested again.

Closing this issue as it is browser related. Thanks!

FernandoGarcia, 2022-04-26:
> You can try this but it is triple compressed

Hi!

Decompress a folder two times is not a problem.

Please, where's the source code used to make this build?

Best regards.