jeelabs/esp-link #548: wifi to rs485

Dinesh-0305, 2022-05-09:
hi all, i used the firmware version alpha 3.2.47,  in that cant able to enable (tx enable) in the pin assignment preset... can anyone suggest idea for this pin to enable to work for wifi to 485

sergedenni, 2022-07-24:
Hi i m interest to do a bridge RS485 ESP 01    ESP 01 RS485 do you know a solution ? i need 115200 baud rate

Freemen16, 2023-09-05:
Hello. "tx enable" doesn't work for me either. It looks like the author abandoned the project. It's a pity.

uzi18, 2023-09-08:
@Freemen16 @sergedenni @Dinesh-0305 there was code to support "tx enable" but  it was reverted some time ago because of bugs.
It's possible to add it one more time if You are ready to test it.

uzi18, 2023-09-08:
patch: https://github.com/jeelabs/esp-link/commit/e57f29c019b1d895d022f2a35e54201ff99df8cc?w=1

alextrical, 2024-01-18:
Partial bug fix, at least to the part that prevented the ESP8266 from booting can be found here
https://github.com/jeelabs/esp-link/issues/489#issuecomment-1898092084