jeelabs/esp-link #552: OTA programming of atmega2560 not working [linux]

MNS26, 2022-08-27:
whenever i try and program a atmega2650 with a esp8266 (esp01s in this case) it first throws 2 `ioctl("TIOCMGET"): Inappropriate ioctl for device` and then times out avrdude

when using the same setup with a atmega328p it works w/o issue...

i need the extra IO of the atmega2560 for this project and cant use usb to program it (will be tucked away behind a lot of stuff)

there have been issues about this before with one claiming to update the bootloader but that didnt work sadly

MNS26, 2022-08-27:
nvm just noticed that the release 3.2.47 has support for atmega2560

closed 2022-08-27 by MNS26

uzi18, 2022-08-29:
@MNS26 you need latest esp-link and good bootloader flashed into atmega2560