jeelabs/esp-link #561: mdns stop working/resolving hostname.local after ~2.5 minutes after boot

rootmax84, 2023-01-03:
Is there any fix of this?

JDP-Maker, 2023-01-04:
On what OS?  On windows 10+ the builtin mdns is pretty strange.  I disabled it and installed bonjour.

rootmax84, 2023-01-05:
> On what OS? On windows 10+ the builtin mdns is pretty strange. I disabled it and installed bonjour.

Any OS. IOS, linux, macos, etc. Bonjour on windows doesn't help.

Jerther, 2024-11-10:
I have the same problem. Fedora Linux 39. Using `avahi-browse -a` I can randomly see it the ESPLink appear (plus sign)

`+ wlp4s0 IPv4 esp-link                                      _arduino._tcp        local` 

but then shortly after, it disappears (minus sign).

`- wlp4s0 IPv4 esp-link                                      _arduino._tcp        local`

it may come back, after a few minutes, or an hour. It seems VERY random. I thought I could make it appear by resetting the hostname setting, but I think it was just a coincidence.

The ONE time I cought it and set it as the programming port in Arduino IDE I got this error:

```
Resetting the board
Failed to reset the board, upload failed
````

It used to work fine though. I'm really not sure what has changed. Last time I used ESPLink to wirelessly program an ATMega in Arduino was more than a year ago. Software updates, security updates, who knows...

I really wish there was an easy way to manually configure the ESP Link parameters in Arduino IDE and not rely on mDNS. Should just be an ip address and a port, right?

rtek1000, 2025-05-04:
I've been using esp-link for years to program the ATMega328 as if it were an "Arduino Uno Wifi", and occasionally the IDE's IP address disappears. But I noticed that if the mDNS service is disabled, esp8266 restarted, service enabled, esp8266 restarted, then the IDE finds the IP address again. I don't always program using esp-link, so this doesn't bother me much.

But it would be very useful if there was an Arduino sketch with this OTA service part using the ESP8266. I know that there is an example of an Arduino OTA sketch in the Arduino IDE, already designed for the ESP8266 to program itself, but it lacks this option to make the ESP8266 an Arduino UNO programmer, the way esp-link does.

https://docs.arduino.cc/arduino-cloud/features/ota-getting-started/