jeelabs/esp-link #563: how to change soft-ap static IP and PORT?

masonlsd, 2023-02-17:
how to change soft-ap static IP and PORT?

I need 192.168.0.10:35000
this is the setting for the obd adapter, car scannet

uzi18, 2023-02-17:
try espeasy with serial server plugin

masonlsd, 2023-02-22:
this will not work, I need to broadcast uart over wifi

masonlsd, 2023-02-22:
can you give me an instruction or a finished firmware file?

uzi18, 2023-02-23:
what exactly module do you want to use and what is your flash size?

masonlsd, 2023-02-24:
I have wemos d1 mini 4MB memory size

uzi18, 2023-02-24:
try collection_C_ESP8266_4M1M

masonlsd, 2023-02-24:
there are plugins:
Communication - Serial Server
Communication - Serial Proxy
Communication - Ping
Communication - P1 Wifi Gateway
Communication - DMX512 TX
setting up the plugin Serial Server, but where do i change the static ip of esp?

uzi18, 2023-02-24:
Serial Server, ip should be settable on one of the first pages.

masonlsd, 2023-02-24:
ok thanks for the help, I will look further, how to change the ip and port

uzi18, 2023-02-24:
https://espeasy.readthedocs.io/en/latest/Config/Config.html#wifi-ethernet-ip-settings