jeelabs/esp-link #564: ESP-link to connect a Smart Meter

ScanxTaz, 2023-02-28:
Hey Dear, hope there's still life here :)

I'm wondering... I succeeded in flashing a NodeMCU with and ESP32 on it with ESP-Link.
The beast is running and I can see the logs of the smart metter rolling on ESP-Link.
Now, I'd like to get the content of the log and parse it to publish on MQTT.
I'm not totally sure ESP-Link can do that... Is it ?

I'm currently trying to bind a socket in Python from an external machine on the port 23 of ESP-Link.
This works, but well... I'm sure it can do it more properly, especially because the socket falls down after a while, seems to have sync issues...

so, is there any more proper way of doing this ?

Thanks for the (alive) people here :)
Thierry

uzi18, 2023-02-28:
esp-link does not support esp32.
NodeMCU is esp8266 thats why it works :)

I'm using telnetlib for communication with esp-link as serial bridge on 23 port

https://github.com/uzi18/sterownik/blob/e53e18d75375e0db319ab2d5652ee19794328e5d/arduino/Domoticz/test.py#L109

ScanxTaz, 2023-03-01:
Yeah, thanks usi18, but either with Telenetlib or direct Coskt implementation, the listener falls at some point. The DSMR is "publishing" the logs of the energy consumption infinitely, and, I don't know why, this implementation fails after a while... (I also started with Telenetlib)

uzi18, 2023-03-01:
Please show more info about your project, maybe we can simulate this behabior?