jeelabs/esp-link #565: No AP

roblomq, 2023-04-09:
I've successfully flashed an ESP-01 but after that there does no AP showup. Tried several times with several versions of the flash download tool tried several ESP-01 boards but all with the same result.
Somebody a tip what I can try?

[image]

uzi18, 2023-04-11:
do you have any logs?

roblomq, 2023-04-12:
No because I can't access it.

uzi18, 2023-04-12:
Use serial port to debug situation.

roblomq, 2023-04-12:
Switched SPI to DOUT and now it works.

Bear1012061, 2023-07-20:
I'm getting it to upload to my 1MB ESP-01, but my ESP-01 won't create a network... Any solutions? The SPI speed is 40MHz and the SPI mode is QIO. Should it say "is stub and send flash finish" when it is finished uploading?

flaviut, 2023-08-14:
I was able to get the flash logs by using `picocom /dev/ttyUSB1 -b 76800`.

I got the following logs:

```
2nd boot version : 1.6
  SPI Speed      : 40MHz
  SPI Mode       : QIO
  SPI Flash Size & Map: 4Mbit(256KB+256KB)
no GPIO select!
jump to run user1 @ 1000

rf_cal[0] !=0x05,is 0xFF

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 2408, room 16 
tail 8
chksum 0xe5
load 0x3ffe8000, len 776, room 0 
tail 8
chksum 0x84
load 0x3ffe8310, len 632, room 0 
tail 8
chksum 0xd8
csum 0xd8
```

I then came across @uzi18's [very helpful comment, and that fixed it](https://github.com/jeelabs/esp-link/issues/543#issuecomment-1090598270):

```
esptool.py -b 1000000 -p /dev/ttyUSB1 write_flash --flash_freq 20m --flash_mode dio --flash_size 4MB 0x1000 user1.bin 0x0000 boot_v1.7.bin 0x3fc000 esp_init_data_default.bin 0x3fe000 blank.bin
```

I did have to download `esp-link-v3.2.47` though, not the default v3.0.