jeelabs/esp-link #568: Issue setting baud rate to 2400

Brynod, 2023-06-04:
Hi all,
I have flashed ESP-Link to my wemos D1 mini and all is working fine on the larger part... however, if i try to change the baud rate to 2400 in the serial console (hoping to use this to montor my RS232 inverter) , this causes issues.

Firmware 2.2.3 simply wont set anything below 9600 and reports "error setting baud rate"
Firmware 3.0.14 will allow the change but then goes into a boot loop after power cycle
firmware 3.2.47 will allow the change but then goes into a boot loop after power cycle

is it possible to use ESP link at such a low baud rate? If so....does anyone know how to do this....or if this is even necessary?  I am waiting for my MAX3232 adaptor to connect the inverter but just wanted to get everything ready in advance.

Thanks in advance!
Regards
Bryan

[image]

Brynod, 2023-06-04:
NB: i have tried setting various different baud rates to gather as much info as possible, the message in the screenshot appears when  trying to set to 2400 from any other baud rate in firmware 2.2.3

Trying this in any later firmware will allow the change but then just bootloops after power cycle and needs full reflash to get working again..

Thanks again  for any help with this
Regards
Bryan

adharyanto, 2026-02-21:
> Code for all baudrates >300 looks the same

Hallo...

I have tried this way; firstly I set the JeeLabs ESPLink ESP8266 to 9600 or 115200, then connect to the Inverter, and of course it will not work because the Inverter works only on 2400. And then I changed the rate from 9600 or 115200 to 2400 (while it is still running and connected to the Inverter)... and it worked, but very slow and a lot of errors on my Inverter application program, (I built it using NodeRED). Then I monitored the stream of serial data (using Serial Monitor application), I could see the streaming data was not what I expected; many of the streaming data lengths were not the consistent, causing my NodeRED application program could not understand the data.

Moreover, after switching off the ESP8266-JeeLabs ESPLink (with the 2400 setting), and tried to switch on back again, it would not work. This means, JeeLabs ESPLink will work on 2400 only after changing from e.g. 9600 to 2400, but when I switch off the JeeLabs ESPLink, when it is set to 2400, and tried to switch on back again, it will not work.
Have you tried to switch off/on your JeeLabs ESPLink ESP8266/32 when it is set to 2400?

I suspect the inconsistent/too many error of the serial data streaming is caused by the conversion from RS232 to TTL, (using MAX3232), maybe the voltage level conversion is not good,.... I think I need to play/modify the capacitors of the MAX3232... just need some efforts & time...

Using the Hi-Flying Elfin EW10 work just fine, all the serial data stream exactly as I expected, and my NodeRED program can read them.

Regards,
Antonius

adharyanto, 2026-02-24:
> [@adharyanto](https://github.com/adharyanto) so maybe there is a bug in hardware of esp8266 and workaround is set first 9600 and next desired baud rate? if you use 32Mbit flash could prepare for you such modified version to test?

Hallo,

If there is a bug, the bug maybe is not in the ESP8266 hardware but maybe in the JeeLabs ESP-Link instead, because ESP8266 is running at 2400 on different software. JeeLabs ESP-Link works at 2400 but only the first time after changing from higher rate, after a power cycle (on/off), JeeLabs ESP-Link will not work/start.

I have tried several other Arduino codes, and testing them at 2400:
- some porting code made by "atadiat", it runs but serial data flow is not stable, my NodeRED program cannot read it
- example code from Arduino ide, esp8266wifi-WifiTelnetToSerial, it runs but again serial data flow is not stable, my NodeRED program cannot read it

I have checked my RS232-TTL voltage conversion, it is not very clean, little noise, but it is ok, data is translated correctly.
The problem i have now is the Flow Control of the serial data stream, i will need the code that has control on the serial data flow (Flow Control), and it must be a software controlled (Xon/Xoff), (not hardware controlled; RTS/CTS/DTR), because the Inverter has only limited communication cables (only VCC, Gnd, Tx, Rx).

So far, only the Hi-Flying Elfin-EW10 (RS232-Wifi module) gives me the solution. It converts the RS232 to WiFi flawlesly, and I can check & monitor the serial data coming very stable, and my NodeRED program can read it.

Regards,
Antonius

adharyanto, 2026-02-26:
> [@adharyanto](https://github.com/adharyanto) You said maybe "esp-link code is with bug", but later you said about 2 arduino codes also works wrong. So conclusion should be opposite - this is like hardware problem. Maybe on ESP32 it is resolved. Will check this w Logic Analyzer on 2400

Hallo,

I have tried 4 codes for my specific case/issue, (Inverter RS232-MAX3232-WiFi(ESP8266/32)-NodeRED, baud rate 2400):

- Code-1 = JeeLabs ESP-Link, using ESP8266
- Code-2 = by Atadiat, using ESP8266
- Code-3 = Arduino IDE, esp8266wifi-WifiTelnetToSerial, using ESP8266
- Code-4 = by Yuri-Rage_ESP-Serial-Bridge, using ESP32

Comparison result:
- Code-1 =
	- problem with baud rate 2400, after power cycle on/off ESP-Link wont work
	- JeeLabs ESP-Link only works on ESP8266 (does not work on ESP32)
	- incoming data is chopped, NodeRED cannot read it
- Code-2 =
	- no problem with baud rate 2400, after power cycle on/off it still works
	- works on ESP8266 and ESP32
	- incoming data is chopped, NodeRED cannot read it
- Code-3 =
	- no problem with baud rate 2400, after power cycle on/off it still works
	- works on ESP8266 and ESP32
	- incoming data is chopped, NodeRED cannot read it
- Code-4 =
	- no problem with baud rate 2400, after power cycle on/off it still works
	- works on ESP8266 and ESP32
	- incoming data is chopped, NodeRED cannot read it

Looking at the above result, we can see that the JeeLabs ESP-Link is the inferior one compared to the other 3 codes.
JeeLabs ESP-Link has problem after power cycle on/off at 2400, the other 3 codes has no problem.
JeeLabs ESP-Link works only on ESP8266 (does not work on ESP32), the other 3 codes works on both ESP8266/ESP32 (use the right library).

So, at baud rate 2400, ONLY using JeeLabs ESP-Link has problem, while using the other 3 codes have no problem.
So, if there is a bug, the bug is NOT on the hardware (ESP8266/ESP32), but MAYBE in the JeeLabs ESP-Link.

Have you tried power cycle on/off using JeeLabs ESP-Link at 2400, and see what happen?

Anyway, the problem for my case (with this specific configuration) is the "Flow Control" of the incoming data, the data is not coming at one piece at one time, but rather it is chopped into several data, this causing my NodeRED fails to understand it.
If I compare to the incoming data from the Hi-Flying Elfin EW10 module, it comes in one piece at one time (not chopped), so my NodeRED can read, analyse, and parse it, to be displayed.

If there are some other codes with "Flow Control" capability, then I would like to try them.
So far, only the Hi-Flying Elfin EW10 gives me the solution, no alternative yet.

Regards,
Antonius

adharyanto, 2026-02-26:
Forgot to mention, I also have tried the Tasmota, Serial/SerBr, the result is the same as the other 3 codes....

uzi18, 2026-02-26:
@adharyanto have checked my build of esp-link (2.4.37) attached as archive in issue #543 (you did not mention firmware version of esp-link)
This one works correct with 115200 and 2400, no crash on reset, on power on etc. but first have disabled debug log on uart via Debug log subpage.
Is it possible to look at your nodered flow configuration?

adharyanto, 2026-02-28:
> [@adharyanto](https://github.com/adharyanto) try this one:
> 
> [esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.zip](https://github.com/user-attachments/files/25609823/esp-link-v3.2.47.alpha.24-g74e2043-dirty-Lucjan.zip)

Hallo,

Great news..... it WORKS now.....!!!
All incoming data is received completely without chopping/partially sent.
Even the parameter data which has 112 chars also coming completely at once.
I do not know if you have modified the JeeLabs ESP-Link from the maximum chars 80 to 115, (or to 120), but for my current NodeRED application it is now working well...

Now I have alternative solution, beside the Hi-Flying Elfin EW10 module.
Thank you very much for your help...
Maybe this update/modification also useful for the others who are facing the same/similar problems.

Regards,
Antonius

uzi18, 2026-02-28:
@adharyanto nice to hear another happy user :)
Will look at it in nodered, but need to install it first, thx. 
With Debug output your tcp in eventually ommited wrong data. Just less data is passed to tcp socket. Not sure you check datagram checksum.

adharyanto, 2026-03-01:
Hallo,

Further observation; looks like I am having some issues:
- measurement result sometimes not updating...
- i already checked; NodeRED keep sending commands/requests to Inverter, (via JeeLabs ESP-Link), but Inverter stop responding or NodeRED does not receive the responses from Inverter...
- further check, commands/requests from NodeRED arrived at JeeLabs ESP-Link, but not passed to Inverter, thus Inverter does not receive the Commands/Requests, and thus does not reply/send the answers...
- this happens after about 20-30 minutes...
- tried to power cycle the JeeLabs ESP-Link, (on-off-on),... then it worked normal again,... but problem happened again after about 20-30 minutes....
- checking on the web based measurement data record (5-minute sampling), as below attached picture, we can see if the "Power Used" (also "Voltage") showing repeated same values, this indicates the measurement is stopped working (red arrows)..... then went back normal again (green arrows)... then stopped again.... and so on... 

Do not know if this because of the WiFi was not stable, of the JeeLabs ESP-Link sometimes frozen and then back normal again..... intermittent.... or something else...??? Still keep monitoring & observing.....

Is there any possibility for JeeLabs ESP-Link to use different port number, instead of port-23?

Regards,
Antonius

[image]

uzi18, 2026-03-02:
E:M 656 means - "error: no mem" 
you can try to use esp-link with debug off and closed esp-link server page

adharyanto, 2026-03-03:
Hallo,

"error: no mem" = Does this mean the ESP-Link is running out of memory? (Because of many commands/requests coming, and causing to freeze for 20 minutes, and then working again for 30 minutes,...?)

I was looking for "esp-link server page" on the esp-link drop down menu, while I did not switch the debug off, then esp-link stop working, no access, and had to re-flash again. I was thinking this "esp-link server page" is part of the menu, but I could not find. I think you mean to close the esp-link webpage/browser.

After re-flash, switch off debug log, change baud rate 2400, close esp-link webpage, and connect again to (NodeRED & Inverter), it works again for about 30 minutes, and then stop working again for about 20 minutes,...... I do not know what is happening here???

Checking the debug log, some errors happened again, and the most suspicious are these:
358624> serbridge: txbuffer full, conn 0x3fff6e3c
370231> serbridge: killing overlowing stuck conn 0x3fff6e3c
Are these causing the periodically "run-30mnt/stop-20mnt"?

Below is part of debug log capture:

M 656
  1012> E:M 656
  1012> E:M 656
  1012> E:M 656
  1012> ERROR! espconn_sent returned -1, trying to send 166 to /console.html
 10976> HTTP GET /console.html: 200, 9981ms, h=14528
 11226> HTTP GET /favicon.ico: 200, 9906ms, h=14832
166611> HTTP GET /home.html: 200, 251ms, h=10376
166901> HTTP GET /ui.js: 200, 323ms, h=10920
166903> HTTP GET /style.css: 200, 327ms, h=11688
167500> HTTP GET /pure.css: 200, 953ms, h=13864
167842> HTTP GET /favicon.ico: 200, 1261ms, h=14960
167938> HTTP GET /menu: 200, 75ms, h=13432
168066> HTTP GET /wifi/info: 200, 128ms, h=11448
168165> HTTP GET /system/info: 200, 228ms, h=12040
169297> HTTP GET /favicon.ico: 200, 1358ms, h=10200
170136> HTTP GET /pins: 200, 2199ms, h=11304
358624> serbridge: txbuffer full, conn 0x3fff6e3c
370231> serbridge: killing overlowing stuck conn 0x3fff6e3c
682673> Accept port 23, conn=0x3fff7e60, pool slot 0
993192> Accept port 23, conn=0x3fff7e60, pool slot 0
220797> HTTP GET /log.html: 200, 5436ms, h=11896
221152> HTTP GET /style.css: 200, 520ms, h=11320
222763> HTTP GET /console.js: 200, 140ms, h=11656
224212> HTTP GET /pure.css: 200, 3588ms, h=13560
224266> HTTP GET /ui.js: 200, 3629ms, h=15008
224654> HTTP GET /menu: 200, 139ms, h=12744
224702> HTTP GET /log/dbg: 200, 185ms, h=12176
224896> HTTP GET /favicon.ico: 200, 375ms, h=13440
233688> HTTP GET /log/text: 200, 9033ms, h=14008

adharyanto, 2026-03-03:
ESP-Link stopped again, check the debug log, showing this:
(looks like "txbuffer full" and "killing overlowing stuck conn ....." causing the problem)....

, h=8080
148120> serbridge: txbuffer full, conn 0x3fff6e58
160116> serbridge: killing overlowing stuck conn 0x3fff6e58
470461> Accept port 23, conn=0x3fff7e28, pool slot 0
780647> Accept port 23, conn=0x3fff7e10, pool slot 0
 91452> Accept port 23, conn=0x3fff7e10, pool slot 0
105884> Accept port 23, conn=0x3fff7f18, pool slot 1
923262> serbridge: txbuffer full, conn 0x3fff6e58
935245> serbridge: killing overlowing stuck conn 0x3fff6e58
246109> Accept port 23, conn=0x3fff7e10, pool slot 0
556314> Accept port 23, conn=0x3fff7e10, pool slot 0
866504> Accept port 23, conn=0x3fff7e10, pool slot 0
935020> Accept port 23, conn=0x3fff7f18, pool slot 1
383617> serbridge: txbuffer full, conn 0x3fff6e58
407934> serbridge: killing overlowing stuck conn 0x3fff6e58
489037> HTTP GET /home.html: 200, 76ms, h=13640
489140> HTTP GET /favicon.ico: 200, 80ms, h=9504
489156> HTTP GET /ui.js: 200, 106ms, h=12136
489160> HTTP GET /pure.css: 200, 122ms, h=13584
489200> HTTP GET /style.css: 200, 159ms, h=15032
489272> HTTP GET /menu: 200, 18ms, h=11016
489279> HTTP GET /wifi/info: 200, 23ms, h=12464
489319> HTTP GET /pins: 200, 64ms, h=13912
489320> HTTP GET /system/info: 200, 63ms, h=14680
493509> HTTP GET /log.html: 200, 86ms, h=15032
493612> HTTP GET /console.js: 200, 89ms, h=15024
493629> HTTP GET /menu: 200, 6ms, h=13904
494188> HTTP GET /log/dbg: 200, 565ms, h=15024

adharyanto, 2026-03-03:
I am thinking of "txbuffer" needs to be increased..... i do not know how many it is now, but maybe needs to be increased by 50%....

adharyanto, 2026-03-03:
Just for your info; the Elfin EW10 UART Buffer size is set to = 512.

adharyanto, 2026-03-05:
> [@adharyanto](https://github.com/adharyanto) maybe could prepare for you test version with 5-6 connections and small buffer 512 bytes per connection ?

Hallo,

Thank you for sending the update of the ESP-Link.
I think this case is a good example as a test-bed for the ESP-Link.

I checked on the Elfin EW10 documentation below, and it is mentioned that by default it can handle up to 5 socket channels, and there is possibility to add more through software configuration.
If this updated ESP-Link has been upgraded to 5-6 channels, maybe it will solve the problem, small buffer size of 512 bytes should be enough to save the memory. I am thinking if ESP-Link has some similar menus/possibilities to reconfigure the parameters (buffer size/socket channel/port number/etc.), or maybe reconfigure via Console Commands.

Let me test this updated ESP-Link, and let me observe for 1-2 hours, and I will let you know the results.

Regards,
Antonius

[image]