jeelabs/esp-link #570: No network created

Bear1012061, 2023-07-20:
I'm getting it to upload to my 1MB ESP-01, but my ESP-01 won't create a network... Any solutions? The SPI speed is 40MHz and the SPI mode is QIO. Should it say "is stub and send flash finish" when it is finished uploading?

Bear1012061, 2023-07-20:
Here are my logs;

*************************** START ****************************
START TIME: 202307192320
CONNECT BAUD: 115200
set state: ESP_DL_SYNC
serial port opened
-----------
baud:115200
root baud:115200
-------------
===============BAUD : 115200===============CALL DEVICE SYNC
connecting...
0x3ff00050: 7dd00000
0x3ff00054: 02005f4c
0x3ff00058: c200b000
0x3ff0005c: 00483fda
check_err_0: 0bcheck_err_1: 00check_err_2: 00check_err_3: 02check_err_4: 0b====================
EFUSE NORMAL MODE
====================
=========================
CRC IN MODE 1: 
crc_calc_res: 194
target crc val: 194
=========================
=========================
CRC IN MODE 1:
crc_calc_res: 208
target crc val: 208
=========================
==================
EFUSE LOG:
======================
    
EFUSE LOG:
    
---------------
    
REG0:7DD00000
    
REG1:02005F4C
    
REG2:c200b000
    
REG3:00483FDA
    
----------------
    
====================
                                   
EFUSE NORMAL MODE
                                   
====================
EFUSE CHECK PASS...48bit mac
-----------------------
 48 3f da 5f 4c 7dCUSTOM_ID: 0C 20 00 00 00 0D 00 00 
MAC AP : 4A-3F-DA-5F-4C-7D
MAC STA: 48-3F-DA-5F-4C-7D
get mac res: 1
get flash id : 0x00144068
manufacturer_id: 0x68
device_id: 0x4014

Wrote 4080 bytes at 0x00000000 in 0.3 seconds (effective 97.7 kbit/s)...
Hash of data verified.
Wrote 344708 bytes at 0x00001000 in 22.1 seconds (effective 125.0 kbit/s)...
Hash of data verified.
Wrote 128 bytes at 0x000fc000 in 0.1 seconds (effective 13.8 kbit/s)...
Hash of data verified.
Wrote 4096 bytes at 0x000fe000 in 0.1 seconds (effective 416.8 kbit/s)...
[... 1058 lines cut ...]
EFUSE CHECK PASS...48bit mac
-----------------------
 48 3f da 5f 4c 7dCUSTOM_ID: 0C 20 00 00 00 0D 00 00 
MAC AP : 4A-3F-DA-5F-4C-7D
MAC STA: 48-3F-DA-5F-4C-7D
get mac res: 1
get flash id : 0x00144068
manufacturer_id: 0x68
device_id: 0x4014

Wrote 3856 bytes at 0x00000000 in 0.3 seconds (effective 97.1 kbit/s)...
Hash of data verified.
Wrote 330756 bytes at 0x00001000 in 21.2 seconds (effective 124.6 kbit/s)...
Hash of data verified.
Wrote 344708 bytes at 0x00081000 in 22.1 seconds (effective 125.0 kbit/s)...
Hash of data verified.
Wrote 128 bytes at 0x000fc000 in 0.0 seconds (effective 23.1 kbit/s)...
Hash of data verified.
Wrote 4096 bytes at 0x000fe000 in 0.1 seconds (effective 418.0 kbit/s)...
Hash of data verified.
Leaving...
com closed
Use time: 45 s
Download result: PASS
*************************** END ****************************

FlorianHeigl, 2024-09-02:
Hi,

if you can, try to flash a 2.x version and verify if it also happens there. I think it won't.