jeelabs/esp-link #573: collect2.exe: error: ld returned 1 exit status

ikalash762, 2023-11-27:
I try to compile this project for ESP8266 (IDE: Visual Studio Code, extension PlatformIO), but i get the following error:

_c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `setup'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\libFrameworkArduino.a(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0xc): undefined reference to `loop'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: .pio\build\esp12e\libFrameworkArduino.a(core_esp8266_main.cpp.o): in function `_ZL12loop_wrapperv':
core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x29): undefined reference to `setup'
c:/users/user/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: core_esp8266_main.cpp:(.text._ZL12loop_wrapperv+0x35): undefined reference to `loop'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp12e\firmware.elf] Error 1_

The content of platformio.ini file is the following:
[env:esp12e]
platform = espressif8266
board = esp12e
framework = arduino
upload_port = COM[3]
upload_speed = 115200
monitor_port = COM[3]

Can you help me to resolve this problem?

uzi18, 2023-11-27:
It is not arduino framework.

ikalash762, 2023-11-28:
Thank you for a quick answer. Please help me a little more, by answering the following questions.
It seems that ESP8266 can work not only with arduino. I have Visual Studio Code + PlatformIO framework (for embedded systems like my ESP8266).
In this case, how can i compile this project? Maybe i can switch it to other platform, which supports this project esp-link (UART to WIFI bridge)? What IDEs and platforms (frameworks) can i use to work with this project on ESP8266?