jeelabs/esp-link #574: RS232 to UDP or TCP

jaygrovr, 2023-12-06:
I dont see  a way to use this code to forward messages received on the uart to a UDP or TCP port?  Am I missing something?  I dont want to use mqtt, i just want it to send it out to a specific port so I can have node-red parse the data.

Thanks

Jason

btw, nice work on what I have so far.

yesyesuk, 2023-12-07:
Hi Jason. Esp-link is a TCP server that waits for incoming connections. You need to use a client to connect to esp-link.
Esp-link is not a TCP client that makes outgoing connections to an IP:port.

Once a client is connected, esp-link will forward data Serial Port <-> TCP in both directions over that TCP connection.

jaygrovr, 2023-12-07:
If my node-red instance makes a connection with it on the network side will I still be able to see data coming in from the serial side?
Jason

yesyesuk, 2023-12-07:
I'm not familiar with node-red (heard the name before, but that's it), but yes, this is how it works in general, and that's how I'm using it (with my own client).

jaygrovr, 2023-12-07:
how do I know which port to request the data from?
jason

yesyesuk, 2023-12-07:
it's port 23

uzi18, 2025-04-14:
or 2323 (but behavior is different)