jeelabs/esp-link #577: wlan bridge/repeater

Vonderhecke, 2024-02-21:
Hello,
I have esp-link running in AP-STA mode via station, WAN can also be reached and a gateway (for static IP) is also specified below under "Special Settings". However, a client logged in via the Soft AP does not have Internet access. It can access both the 192.168.4.1 and the station IP, for me a static one. So on the esp-link web server and Telnet but not on the Internet. Would it be possible?
Many thanks for the help.

closed 2024-03-12 by Vonderhecke

uzi18, 2024-03-12:
Don't get it, what do you mean by "not on internet"?

Vonderhecke, 2024-03-12:
It's a shame, but it was just an idea that he could have expanded the WiFi a little in addition to the actual tasks.

uzi18, 2024-03-13:
Not sure if it is possible, should dig some information from IDF documentation.