jeelabs/esp-link #585: Not flashing on an NodeMcu ESP32 wroom

5p0ng3b0b, 2025-02-01:
Hi, I have tried to flash the latest firmware in linux with the following command:
`esptool.py --port /dev/ttyUSB0 --baud 921600 --chip esp32 write_flash 0x1000 user1.bin 0x3FC000 esp_init_data_default.bin 0x3FE000 blank.bin 0x00000 boot_v1.6.bin`

It does not work? Here is the output from running it.

```
esptool.py v4.8.1
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 94:e6:86:02:a3:50
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Unexpected chip id in image. Expected 0 but value was 2408. Is this image for a different chip model?

A fatal error occurred: boot_v1.6.bin is not an ESP32 image. Use --force to flash anyway.
```
If I omit the boot_v1.6.bin file, the other files flash OK. Can I use the --force option and if so how do I use it in the supplied command? Is there a working alternative boot image for this device to write instead? I have written this esp-link firmware to the old micro usb NodeMcu devices in the past and I am unable to recall experiencing any issues there.

5p0ng3b0b, 2025-02-01:
I got impatient and added --force to the end of the command (use the force lol). Everything flashed fine and I can still reflash. I do not however see an esp-link ssid in my wifi settings to connect and configure it though.

uzi18, 2025-02-01:
It is not compatible with esp32

5p0ng3b0b, 2025-02-03:
It is an esp32 afaik. Bought it a year ago but been busy and forgetful. It is an ESP-WROOM-DA MODULE as far as Arduino IDE is concerned and I can flash WiFiTelnetToSerial on it. I tried with the flaky Expressiv windows flash app from the SDK. After flashing and booting, I monitor the com port @115200 and get:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x000004ea
invalid header: 0x000004ea
invalid header: 0x000004ea
invalid header: 0x000004ea
invalid header: 0x000004ea
invalid header: 0x000004ea
invalid header: 0x000004ea
invalid header: 0x000004ea
ets Jul 29 2019 12:21:46
```
I tried another windows app from here https://github.com/marcelstoer/nodemcu-pyflasher (I had to make a combined fw first). It would not flash due to the following error.
```Unexpected chip id in image. Expected 0 but value was 2592. Is this image for a different chip model?```

drencorxeen, 2025-02-03:
@5p0ng3b0b 
ESP-Link is for the ESP-8266 modules.  If you look at the main README is says to prepair  your ESP-8266.

Based on this, you will need a ESP-8266 based module not a ESP-32 based module.

uzi18, 2025-02-03:
@5p0ng3b0b have ignorred all the warnings about possible wrong chip...

5p0ng3b0b, 2025-02-04:
Oops, i'm such a bell-end, I gave my brother the 8266 by mistake then. I wondered why I'd flashed before without issue. I'll have to swap it with him. Thank you, I appreciate your help and apologies for my stupidity. I have also learned that not all ESPs are equal. :)

drencorxeen, 2025-02-05:
@5p0ng3b0b 

Not a problem.  I think at some point we all do things like that.  Why there are phrases like, "If it was under your nose it would have bit you." or other type phrases.

I bought a few of the ESP8266-01 modules at Micro Center.  I am sure there are other model of dev boards that have more control  lines for you to use as well.

Hope you find a model that works for you.

uzi18, 2025-02-06:
@5p0ng3b0b @drencorxeen  In fact there is still demand for esp32 port  :)