jeelabs/esp-link #589: swap the serial pins

rew1ndustry, 2025-05-20:
my world is all wemos d1 mini boards, and these have UART0 wired to the USB transceiver.

is it possible to swap serial to the alternate pins?

uzi18, 2025-05-20:
Yes it is but not as software serial.

rew1ndustry, 2025-05-20:
i do not understand - i have no use for software serial, that i know of - does esp-link not use hardware uart0 on the 8266?

on wemos d1 mini boards, the normal uart0 pins (1 and 3) are hardwired to a USB chip, and it is necessary to move uart0 to the auxiliary pins (13 and 15) in order to get a clean serial connection.  normally 13 and 15 are RTS and CTS, but these are not needed, in my application.

does esp-link have the ability to switch to these alternate pins?

uzi18, 2025-05-20:
Yes as wrote you. Configuration is on selfhosted webserver

uzi18, 2025-05-20:
uart swapped option:
https://cloud.githubusercontent.com/assets/39480/19334029/f8128c92-90ad-11e6-804e-9a4796035e9a.png

uzi18, 2025-05-20:
@rew1ndustry try full backup from here: https://github.com/uzi18/sterownik/tree/master/arduino/esp-link
apmode Lucjan - to flash at 0x0

rew1ndustry, 2025-05-24:
thank you everyone, i was wasting your time - should have just installed, and looked - it is all there.

truly excellent firmware this - very much appreciated.

closed 2025-05-24 by rew1ndustry